tensorflow/tflite-micro #119: Project generation fail, missing gemmlowp

yorange1, 2021-06-03:
System information
Ubuntu 20.04

I want to use Project generation in TFLM to output a tree containing only the sources and headers needed to use TFLM for a specific configuration. And my steps are as follows
**STEP1:**
`git clone https://github.com/tensorflow/tensorflow.git`
**STEP2:**
`cd tensorflow`
**STEP3:**
`python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py --makefile_options="OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET_ARCH=cortex-m4" "project"`

The execution of **STEP3** fails, and the output is as follows:
```
Traceback (most recent call last):
  File "tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py", line 193, in <module>
    _copy(src_files, dest_files)
  File "tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py", line 105, in _copy
    shutil.copy(src, dst)
  File "/usr/lib/python3.8/shutil.py", line 415, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.8/shutil.py", line 261, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: 'tensorflow/lite/micro/tools/make/downloads/gemmlowp/fixedpoint/fixedpoint.h'
```

And I found that this problem is caused by **not downloading third_party code**，the list_third_party_* rules will output the path of some undownloaded third_party files,  causing error in project_generation copy function.

I think we can download third_party code before executing the list_third_party_* rules to avoid this problem.

yorange1, 2021-06-04:
@advaitjain

advaitjain, 2021-06-04:
Right, that is the reason why we are currently doing the following:
https://github.com/tensorflow/tflite-micro/blob/c23ed9c4d88482342f8978b2ae67b71c31b152f7/.github/workflows/ci.yml#L90-L93

Corresponding internal bug: http://b/184886633

I'll create a PR to fix this.

yorange1, 2021-06-04:
Thanks for reply，I have submitted some code to try to fix this problem, maybe I can create a PR, and then you can review whether the code is correct

advaitjain, 2021-06-04:
In this case, I went ahead and made a PR for this. Please let me know if it does not fix your issue: https://github.com/tensorflow/tflite-micro/pull/129

And please feel free to submit PRs for bug fixes -- we welcome such contributions.

yorange1, 2021-06-04:
Thanks, I have seen the PR you created, I will try it after the PR is merged

closed 2021-06-07 by advaitjain

yorange1, 2021-06-08:
> Thanks, I have seen the PR you created, I will try it after the PR is merged

Great! I think it fix my issue.

PR #129 (2021-06-07): Have the project generation script explicitly call make third_party_downloads.
This is required only because some of the third_party_downloads are still using the Makefile functions and thus require an explicit `make third_party_downloads`.

Additionally, we sometimes also want to avoid all downloads and this PR adds in a flag for that as well.

Fixes #119 and the corresponding internal bug http://b/184886633

Related bugs are http://b/143904317 and http://b/183229719
commit ce5865c4 (2021-06-07): Have the project generation script explicitly call make third_party_downloads. (#129)

This is required only because some of the third_party_downloads are
still using the Makefile functions and thus require an explicit
`make third_party_downloads`.

Additionally, we sometimes also want to avoid all downloads and this PR
adds in a flag for that as well.

Fixes #119 and the corresponding internal bug http://b/184886633

Related bugs are http://b/143904317 and http://b/183229719