tensorflow/tflite-micro #120: Segmentation fault when running micro_features_generator_test

jenselofsson, 2021-06-03:
@tensorflow/micro

**System information**
- Host OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Ubuntu 18.04
- TensorFlow installed from (source or binary): source
- Tensorflow version (commit SHA if source): b4a116f17909d5fc0360133aa49ce1bddcddc8e4
- Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.): x86

**Describe the problem**
Running `make -f tensorflow/lite/micro/tools/make/Makefile test` results in segmentation fault in ```micro_features_generator_test```.
```
Testing TestMicroFeaturesGeneratorYes
tensorflow/lite/micro/examples/micro_speech/Makefile.inc:259: recipe for target 'test_micro_features_generator_test' failed
make: *** [test_micro_features_generator_test] Segmentation fault (core dumped)
```

**Please provide the exact sequence of commands/steps when you ran into the problem**
```
$ make -f tensorflow/lite/micro/tools/make/Makefile BUILD_TYPE=debug test_micro_features_generator_test
```

Running the binary in ```gdb``` produces the following stacktrace:
```
$ gdb --quiet tensorflow/lite/micro/tools/make/gen/linux_x86_64_debug/bin/micro_features_generator_test 
Reading symbols from tensorflow/lite/micro/tools/make/gen/linux_x86_64_debug/bin/micro_features_generator_test...done.
(gdb) r
Starting program: tensorflow/lite/micro/tools/make/gen/linux_x86_64_debug/bin/micro_features_generator_test 
Testing TestMicroFeaturesGeneratorYes

Program received signal SIGSEGV, Segmentation fault.
0x0000555555556da4 in CalculateCenterFrequencies (center_frequencies=0x5555557605e6 <fixed_pool+9382>, upper_frequency_limit=7500, lower_frequency_limit=<optimized out>, num_channels=41) at tensorflow/lite/experimental/microfrontend/lib/filterbank_util.c:49
49          center_frequencies[i] = mel_low + (mel_spacing * (i + 1));
(gdb) backtrace
#0  0x0000555555556da4 in CalculateCenterFrequencies (center_frequencies=0x5555557605e6 <fixed_pool+9382>, upper_frequency_limit=7500, lower_frequency_limit=<optimized out>, num_channels=41) at tensorflow/lite/experimental/microfrontend/lib/filterbank_util.c:49
#1  FilterbankPopulateState (config=config@entry=0x7fffffffd3f0, state=state@entry=0x55555575e0a8 <(anonymous namespace)::g_micro_features_state+104>, sample_rate=sample_rate@entry=16000, spectrum_size=257) at tensorflow/lite/experimental/microfrontend/lib/filterbank_util.c:97
#2  0x00005555555578b8 in FrontendPopulateState (config=config@entry=0x7fffffffd3e0, state=state@entry=0x55555575e040 <(anonymous namespace)::g_micro_features_state>, sample_rate=sample_rate@entry=16000) at tensorflow/lite/experimental/microfrontend/lib/frontend_util.c:45
#3  0x0000555555554f9b in InitializeMicroFeatures (error_reporter=0x7fffffffd450) at tensorflow/lite/micro/examples/micro_speech/micro_features/micro_features_generator.cc:53
#4  0x00005555555549b2 in main (argc=<optimized out>, argv=<optimized out>) at tensorflow/lite/micro/examples/micro_speech/micro_features/micro_features_generator_test.cc:34
```

The same problem also occurs for ```test_feature_provider_mock_test``` and ```test_feature_provider_test```

advaitjain, 2021-06-03:
Can you try a clean build? I'm not able to reproduce this error locally and the CI is also not catching it.

Assigning to @petewarden

advaitjain, 2021-06-04:
Can you try with https://github.com/tensorflow/tflite-micro/pull/127?

Might be that the underlying issue from https://github.com/tensorflow/tflite-micro/issues/123 is making the test on x86 flaky.

jenselofsson, 2021-06-10:
Sorry for the delayed response, I was certain that I had already responded but perhaps I forgot to actually submit my comment.
About a week ago I tried both with and without #127, and with #127 building worked fine.

I was using gcc version 7.5, and when I switched to using gcc version 8.4, it worked both with and without #127.

I just tried building (```make -f tensorflow/lite/micro/tools/make/Makefile test```) the main-branch with both gcc-7.5 and gcc-8, and it works fine in both cases. So from my end, we can close the issues.

closed 2021-06-10 by jenselofsson