tensorflow/tflite-micro #122: Xtensa optimized kernels for add and mul

advaitjain, 2021-06-03:
As noted in https://github.com/tensorflow/tflite-micro/pull/115#pullrequestreview-674750771 we will need some refactoring prior to adding in optimized implementations for add and mul for Xtensa.

This will include refactoring the reference and cmsis_nn kernels and then adding the Xtensa kernels with optimized calls for Hifi5 and reference fallback for all other target architectures.

advaitjain, 2021-06-03:
tagging @bhanuprakashbv

bhanuprakashbv, 2021-06-04:
@advaitjain, please let me know is there any action item for me related to refactoring the reference kernels.
If so, we can discuss via email.

Along with mul, add activations operators, I am planning to raise pull request for other hifi5 operators as well.
Do we need to refactor for these reference kernels as well?

PR #348 (2021-10-01): REF_CODE_REFACTOR: Add, Mul and Sub
Refactoring the reference code for Add, Mul and Sub operators.

BUG=Fixes #122
closed 2021-10-01 by advaitjain