tensorflow/tflite-micro #123: micro_features_generator_test failing for the Xtensa target

advaitjain, 2021-06-03:
The following command:
```
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=xtensa TARGET_ARCH=fusion_f1 OPTIMIZED_KERNEL_DIR=xtensa XTENSA_CORE=F1_190305_swupgrade test_micro_features_generator_test -j8
```

Fails with:
```
Testing TestMicroFeaturesGeneratorYes
*WARNING* Unhandled user exception: LoadStoreAlignmentCause (0x000130da)( Xtensa ISS ) *WARNING* run exited with status 'virtual breakpoint'
```

The immediate workaround to get the Xtensa build to pass again will be to disable this failing test.

advaitjain, 2021-06-03:
tagging @bhanuprakashbv @nyadla-sys

advaitjain, 2021-06-03:
Note that this test was likely broken by: https://github.com/tensorflow/tensorflow/pull/47935

advaitjain, 2021-06-03:
feature_provider_test and feature_provider_mock_test are also failing for Hifi5 and Vision P6.

bhanuprakashbv, 2021-06-04:
@vp-cad

nyadla-sys, 2021-06-04:
@Sjoshi768

closed 2021-06-19 by advaitjain