tensorflow/tflite-micro #137: Group Convolution Support

will-rice, 2021-06-07:
Any plan on supporting group convolutions in this repo? I would be interested in working on this also. Just wanted to verify it wasn't already in progress before I attempted it.

advaitjain, 2021-06-07:
In general, the recommendation is to first implement the feature in TfLite and then port to TFLM for embedded use-cases.

I see that you have found and commented on https://github.com/tensorflow/tensorflow/issues/40044 and I would encourage you to follow up there for now.

closed 2021-06-08 by will-rice

HemanthSai7, 2024-09-16:
Hello, are the grouped convolutions supported yet? In the TF issue, it seems to have been added.

mansnils, 2024-09-19:
Hello @HemanthSai7,
At least the CMSIS-NN kernel supports it: https://github.com/tensorflow/tflite-micro/blob/6147f20441aaa0be00920a407dacd06a44364188/tensorflow/lite/micro/kernels/cmsis_nn/conv.cc#L94

HemanthSai7, 2024-09-19:
I'm using conv1d in esp32 and I don't think it's there.