tensorflow/tflite-micro #149: Problems to register a custom op in TF Lite

Lucy20211, 2021-06-09:
**System information**

 -   OS Platform and Distribution: Linux Ubuntu 20.10
 -  TensorFlow version: 2.4.1
 -   Python version : 3.8 (installed via pip)
 -   STM32CubeIDE-Lnx: 1.3.0
 -   STM32 Nucleo-64 development board with STM32F401RE MCU

**Describe the problem**

I'd like to forecast RMSE values for seven days from a household power consumption dataset, using seven input test values corresponding to certain power consumption values, with a model which uses a custom op (Conv1D) and two builtin ops (Reshape and Fully Connected). I'm using a STM32 Nucleo-64 development board. The code is the following:

      #include "main.h"
      #include <string.h>
     
      #include "tensorflow/lite/micro/kernels/micro_ops.h"
      #include "tensorflow/lite/micro/micro_error_reporter.h"
      #include "tensorflow/lite/micro/micro_interpreter.h"
      #include "tensorflow/lite/micro/micro_mutable_op_resolver.h"
      #include "tensorflow/lite/version.h"
      #include "tensorflow/lite/schema/schema_generated.h"
      #include "power_consumption.h"
      #include "tensorflow/lite/micro/micro_allocator.h"
      #include "tensorflow/lite/micro/micro_op_resolver.h"
      #include "tensorflow/lite/core/api/flatbuffer_conversions.h"
      #include "flatbuffers/flatbuffers.h"  // from @flatbuffers
      #include "tensorflow/lite/c/builtin_op_data.h"
      #include "tensorflow/lite/c/common.h"
      #include "tensorflow/lite/core/api/error_reporter.h"
      #include "tensorflow/lite/kernels/internal/compatibility.h"
      #include "tensorflow/lite/kernels/internal/reference/conv_1d.h"
      #include "tensorflow/lite/schema/schema_generated.h"

      CRC_HandleTypeDef hcrc;
      TIM_HandleTypeDef htim11;
      UART_HandleTypeDef huart2;
      
      // TFLite globals
      namespace {
      tflite::ErrorReporter* error_reporter = nullptr;
      const tflite::Model* model = nullptr;
      tflite::MicroInterpreter* interpreter = nullptr;
      TfLiteTensor* model_input = nullptr;
      TfLiteTensor* model_output = nullptr;
      constexpr int kTensorArenaSize = 2 * 1024;
      __attribute__((aligned(16)))uint8_t tensor_arena[kTensorArenaSize];
      } // namespace

      void SystemClock_Config(void);
      static void MX_GPIO_Init(void);
      static void MX_USART2_UART_Init(void);
      static void MX_CRC_Init(void);
      static void MX_TIM11_Init(void);

      int main(void)
      {

        char buf[100];
        int buf_len = 0;
        TfLiteStatus tflite_status;
        uint32_t num_elements;
        uint32_t timestamp;
        float y_val[7];
        int i;

        HAL_Init();
        SystemClock_Config();
        MX_GPIO_Init();
        MX_USART2_UART_Init();
        MX_CRC_Init();
[... 228 lines cut ...]
I think the problem is also not having register.cpp and register_ref.cpp files (I don't find them in tensorflow/lite/kernels), so I don't know if I can create these files myself or not (in this case I should be careful with the various header files).

In the BUILD file (I don't even have that) I'd write like this:

    cc_library(
        name = "builtin_op_kernels",
        srcs = BUILTIN_KERNEL_SRCS + [
           "conv_1d.cc",
         ], 
         hdrs = [
            "dequantize.h",
            "conv_1d.h",
         ],
         copts = tflite_copts() + tf_opts_nortti_if_android() + EXTRA_EIGEN_COPTS,
         visibility = ["//visibility:private"],
         deps = BUILTIN_KERNEL_DEPS + [
         "@ruy//ruy/profiler:instrumentation",
         "//tensorflow/lite/kernels/internal:cppmath",
         "//tensorflow/lite:string",
         "@farmhash_archive//:farmhash",
         ],
         )

I'm not sure if I'm missing some other steps to register the custom op.
I used this tutorial as a reference: https://www.digikey.com/en/maker/projects/tinyml-getting-started-with-tensorflow-lite-for-microcontrollers/c0cdd850f5004b098d263400aa294023.

advaitjain, 2021-06-09:
Answering this in more general terms first.

Our default recommendation would be to use Conv2d: https://github.com/tensorflow/tensorflow/issues/43141#issuecomment-813046876

However, as you have mentioned in https://github.com/tensorflow/tensorflow/issues/48290#issuecomment-816123176, the path forward for you is to have a custom kernel.

Since you are using TFLM (instead of TfLite) all the kernels (custom or builtin) need to be in the https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/kernels directory.

Can you try to use the pattern of the circular_buffer custom TFLM op:

 * https://github.com/tensorflow/tflite-micro/blob/0afb62f9d1074df11c92f259a2eb1fc936009b96/tensorflow/lite/micro/kernels/circular_buffer.cc

 * https://github.com/tensorflow/tflite-micro/blob/0afb62f9d1074df11c92f259a2eb1fc936009b96/tensorflow/lite/micro/kernels/micro_ops.h#L74

 * https://github.com/tensorflow/tflite-micro/blob/0afb62f9d1074df11c92f259a2eb1fc936009b96/tensorflow/lite/micro/micro_mutable_op_resolver.h#L161-L164

 * https://github.com/tensorflow/tflite-micro/blob/0afb62f9d1074df11c92f259a2eb1fc936009b96/tensorflow/lite/micro/kernels/BUILD#L275

Lucy20211, 2021-06-09:
Thanks for your reply: I modified the files above for my case.

Is it correct to give a name to Conv1D in this way, with name=...:

    model.add(Conv1D(filters=16, kernel_size=3, activation='relu', input_shape=(n_timesteps,n_features), name="cd1"))

and add neural network layer op as follows?

     static tflite::MicroMutableOpResolver<3> micro_op_resolver;
          tflite_status = micro_op_resolver.AddCustom(
          "cd1", tflite::ops::micro::Register_CONV_1D());

          if (tflite_status != kTfLiteOk)
          {
            error_reporter->Report("Could not add Conv op");
            while(1);
          }

I can still compile it, but when I open PuTTY, I can read anything. If what is written in this last comment is right, then I would try to review the implementation of my custom op.

advaitjain, 2021-06-09:
Unfortunately I do not have any out of the box solutions for you. In my opinion, your best bet is to see if you can use conv2d during training.

If you continue down the path of a custom op, what you need to ensure is that the tflite model has the same custom op as well. This is going to be tricky in your case because converting from TF to tflite is going to switch out conv1d with expand_dims and conv2d.

What you could to do here is to modify the flatbuffer after conversion to add in your custom op instead of the expand_dims + conv2d. While technically possible, this is off the beaten path so you will be on your own w.r.t. working with flatbuffers.

For completeness, I wanted to call out that porting https://github.com/tensorflow/tensorflow/blob/4b6c1e5418c43c38d3b0f9e3bd45e4c915439a1a/tensorflow/lite/kernels/expand_dims.cc from TfLite to TFLM (e.g. picking up https://github.com/tensorflow/tensorflow/pull/35189) would not really help since Conv2d does not support causal padding (https://github.com/tensorflow/tensorflow/issues/48567#issuecomment-827096161).

The underlying issue is the support for causal padding in both TfLite and TfLite Micro https://github.com/tensorflow/tensorflow/issues/48567.

To echo my comment on https://github.com/tensorflow/tflite-micro/issues/137:
When there is a feature that is missing in both TfLite and TFLM, the path forward is to first implement in TfLite and then port to TFLM.

Lucy20211, 2021-06-10:
Sorry, I've never modified a flatbuffer after conversion before. Should I modify the flatbuffer to replace expand_dims and convd2 with the conv1d and convert the new tflite model to a constant C array I can include in my firmware program, then add a file.cc of my custom op in tensorflow/lite/kernels and tensorflow/lite/micro/kernels? Am I wrong and do I also need to perform other steps?

Lucy20211, 2021-07-12:
Sorry @advaitjain,
is it possible to create a custom Conv1D layer by following this link: https://www.tensorflow.org/guide/keras/custom_layers_and_models, then try to implement the corresponding TF Lite op (by following https://www.tensorflow.org/lite/guide/ops_custom) instead?

petewarden, 2021-07-26:
That route may be possible, but unfortunately it's not a path we have any documentation or support for, since it's a pretty advanced customization. As @advaitjain says above, we recommend using Conv2D with a 1xSize kernel shape instead of Conv1D. Closing this bug, unless there's some information I'm missing?

closed 2021-07-26 by petewarden