tensorflow/tflite-micro #159: shufflenet channel shuffle error

zhangkui669, 2021-06-10:
I use shufflenetv2, the channel_shuffle unit is:
```
_, height, width, channels = inputs.shape.as_list()  
channels_per_group = channels  // groups
# reshape  
x = layers.Reshape((height, width], groups, channels_per_group))(inputs)  
# transpose
x = layers.Permute((1,2,4,3))(x)
# flatten
x = layers.Reshape((height, width, channels ))(x)
```

But this operation:
```
_, height, width, channels = inputs.shape.as_list()
```
in tflite model is regard as Shape and StridedSlice operation, StridedSlice FP32/INT8/UINT8 datatype, so this node will failed when invoke.

How can i fix it?

zhangkui669, 2021-06-10:
[image]
the shape and stridedslice operation caused by Reshape

zhangkui669, 2021-06-10:
same as Upsample2D operation, this op will be decomposed as (Shape, Stridedslice, Mul, ResizeBilinear)， the Mul op didnot support int32 input.

petewarden, 2021-07-26:
Can you describe what data types need to be supported in which ops? With specific bugs on each (for example Mul needing int32 x int32 support) we can keep better track of the requests. If you want to drop an email to petewarden@google.com describing your use case, that will help us with prioritization.

gitE0Z9, 2021-07-31:
Hi, I used this solution as a workaround. It simply adds int32 case and converts to int8.
Not the best method but work. 
https://stackoverflow.com/questions/64850356/error-with-strided-slice-in-tensorflow-lite

njeffrie, 2021-08-04:
If you do what the stack overflow answer says, I would suggest using int32  rather than int8, since GetTensorData<int8_t>() will interpret the input data incorrectly. It looks like there are int32 reference implementations of Mul and StridedSlice in TFLite, so using GetTensorData<int32_t>() should work OK.

We have noted the request for int32 support in these operators, but it's likely that official support is not going to happen for a while.

closed 2021-08-04 by njeffrie

TianpengBu, 2022-01-25:
> 

You can solve the problem by re-formulating the channel shuffle operation.
Instead of shuffling the tensor itself, shuffling the channel index of the tensor, and finally you slice the tensor by using the shuffled index. 

Code below: 
[image]

This will also speed up the inference on tflite.