tensorflow/tflite-micro #161: Question about image conversion in person detection example

wuruoyu, 2021-06-10:
Hi,

I want to use my own image to test [person detection](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/person_detection) and I follow the instructions here, but the byte array I get is too long. 
https://github.com/tensorflow/tflite-micro/blob/f583f92992c3c9dfb8e10f36d66b2fe7267cf7bc/tensorflow/lite/micro/examples/person_detection/person_image_data.h#L17-L20

For instance, I test it with my Github avatar:
[image]

Here are the instructions I use to generate C array file
1. convert -colorspace Gray avatar.png avatar_g.png (to convert it to grayscale img)
2. convert -resize 96x96\! avatar_g.png avatar.bmp3
3. xxd -s 54 -i avatar.bmp3 > avatar.cc

The result C array is of length 27648 (96 * 96 * 3) I do not figure out how to convert it to single-channel since the expected input is (96 * 96). I tried but failed. Could you help take a look at this and update the instructions? Thanks in advance!

alanvgreen, 2021-06-18:
You might try asking on an imagemagick forum. The exact sequence of commands is dependent on the version of imagemagick (the "convert" command) that you are using.

petewarden, 2021-07-26:
Closing, since I believe this is outside the scope of the project.

closed 2021-07-26 by petewarden