tensorflow/tflite-micro #164: Casting int32_t* to int* in call to  tflite::PopulateConvolutionQuantizationParams

alanvgreen, 2021-06-11:
All calls to  tflite::PopulateConvolutionQuantizationParams seem to cast the per_channel_output_shift param from int32_t* to int*.

For example https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/depthwise_conv_common.cc#L116  

This is not causing problems currently, but I expect that it may be a source of future bugs.

advaitjain, 2021-06-11:
Yeah, I have been puzzled by this in the past as well. The not completely satisfying answer is that TfLite and TFLM currently implicitly assume that ints are 32 bit.

In this particular case we have two different functions that expect per_channel_output_shift to be an int* and an int32_t*

https://github.com/tensorflow/tflite-micro/blob/68dd0cf4ff9b46f8ff7e0fb2a416714fb0ba0f2b/tensorflow/lite/micro/kernels/depthwise_conv.cc#L71-L73

https://github.com/tensorflow/tflite-micro/blob/68dd0cf4ff9b46f8ff7e0fb2a416714fb0ba0f2b/tensorflow/lite/kernels/internal/reference/integer_ops/depthwise_conv.h#L22-L24

https://github.com/tensorflow/tflite-micro/blob/f583f92992c3c9dfb8e10f36d66b2fe7267cf7bc/tensorflow/lite/micro/kernels/depthwise_conv_common.cc#L111-L117

https://github.com/tensorflow/tflite-micro/blob/68dd0cf4ff9b46f8ff7e0fb2a416714fb0ba0f2b/tensorflow/lite/kernels/kernel_util.cc#L210-L215

alanvgreen, 2021-06-12:
I suppose 32 bit integers are a fairly safe assumption, but it would be nice to make the type names consistent.

Would you be open to a patch to make the PopulateConvolutionQuantizationParams() per_channel_shift parameter an int32_t* , and remove the reinterpret_cast<>s ?

advaitjain, 2021-06-12:
Yeah, that's going to help. The change to PopulateConvolutionQuantizationParams will have to go in the tensorflow/tensorflow repo and it will then get synced to the current repo automatically.

commit dc276954 (2021-06-14): Manually pull in TFLM changes from b050465c1de2d2d633655a83e504d8fbf86edb9b

These are necessary to have all the required checks pass and will fix #164
commit a0a55b5a (2021-06-14): Automated sync from github.com/tensorflow/tensorflow (#172)

* Sync from upstream TF.

* Manually pull in TFLM changes from b050465c1de2d2d633655a83e504d8fbf86edb9b

These are necessary to have all the required checks pass and will fix #164
closed 2021-06-14 by advaitjain