tensorflow/tflite-micro #170: Trouble using Make to build examples.

embeddetech, 2021-06-13:
I'm seeing build errors with the different TFLM build attempts I've been making. I have not tried a Bazel build, if there are build instructions on how to do a Bazel build for this repo on Windows I would try that, currently I am trying to build with a Make project. I am successfully building Tensorflow Lite (from the main Tensorflow repo) with CMake, but with that build I don't have clarity on what is just needed for TFLM.

Build attempt #1:

Following the first Micro Speech example "Deploy to ARC EM SDP" found here:
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech

Command line commands from an empty folder:
git clone https://github.com/tensorflow/tflite-micro tflite-micro
cd tflite-micro
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=arc_emsdp ARC_TAGS=reduce_codesize OPTIMIZED_KERNEL_DIR=arc_mli generate_micro_speech_mock_make_project

Result: Several packages are downloaded, but the end result is "make: *** No rule to make target 'generate_micro_speech_mock_make_project'.  Stop."

Build attempt #2:

Following the  Micro Speech example "Deploy to ESP32" found here:
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech

ESP IDF should be properly installed.

Command line commands from an empty folder:
git clone https://github.com/tensorflow/tflite-micro tflite-micro
cd tflite-micro
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp generate_micro_speech_esp_project

Result:
There are a couple "FIND: Parameter format not correct" messages and "File not found - *.cc", "File not found - *.h". It appears that pigweed is downloaded, along with some other packages. Ultimately, the build fails with "make: *** No rule to make target 'generate_micro_speech_esp_project'.  Stop.".

Build attempt #3:

Following the Micro Speech example "Deploy to Sparkfun Edge" found here:
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech

Command line commands from an empty folder:
git clone https://github.com/tensorflow/tflite-micro tflite-micro
cd tflite-micro
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge OPTIMIZED_KERNEL_DIR=cmsis_nn micro_speech_bin

Results: Essentially the same results as build attempt #2 above.

Build attempt #4:

Following the Micro Speech example "Deploy to STM32F746" found here:
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/micro_speech

Command line commands from an empty folder:
git clone https://github.com/tensorflow/tflite-micro tflite-micro
cd tflite-micro
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=disco_f746ng OPTIMIZED_KERNEL_DIR=cmsis_nn generate_micro_speech_mbed_project

Same results as above.

Build attempt #5:

Following the Hello World example "Deploy to SparkFun Edge" found here:
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/hello_world

Command line commands from an empty folder:
git clone https://github.com/tensorflow/tflite-micro tflite-micro
cd tflite-micro
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge hello_world_bin

Same results as above.

advaitjain, 2021-06-14:
Windows is currently fully community supported and the builds are not part of any CI or something that the TFLM engineers are set up to test. Often errors are the result of some incompatibility with the shell commands (which appears to be the case here).

Bazel is linux only and TFLM does not have any support for cmake at this time.

I would recommend attempting to build the reference kernels and pasting the full error log from the following command:
```
make -f tensorflow/lite/micro/tools/make/Makefile hello_world
```

Also, if the final goal is to use TFLM from within an IDE, I would suggest trying out the project generation script. Something like (copied from https://github.com/tensorflow/tflite-micro/blob/16a9b1e8bfd69a73fcadff210aecf63085d9cbd4/tensorflow/lite/micro/tools/ci_build/test_project_generation.sh):

```
git clone tensorflow/tflite-micro tflite-micro
cd tflite-micro
python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py /tmp/tflm-out -e hello_world
cp tensorflow/lite/micro/tools/project_generation/Makefile /tmp/tflm-out
cd /tmp/tflm-out
make -j8 examples
```

If you can get these steps to work on Windows, you can also use the tree  from /tmp/tflm-out in your IDE.

embeddetech, 2021-06-14:
Thank you! I am attaching the errors for the two recommendations in "make hello_world errors.txt" (the first step suggested) and "python create_tflm_tree errors.txt". The python tree build might work if the first one worked? 

I didn't know about the python tree build script, that could be helpful. Hopefully I can fix this hello_world make build and perhaps get some traction. 

[make hello_world errors.txt](https://github.com/tensorflow/tflite-micro/files/6650702/make.hello_world.errors.txt)

[python create_tflm_tree errors.txt](https://github.com/tensorflow/tflite-micro/files/6650705/python.create_tflm_tree.errors.txt)

advaitjain, 2021-06-14:
The error that you are seeing is:
```
FIND: Parameter format not correct
FIND: Parameter format not correct
File not found - *.cc
File not found - *.h
```

possibly coming from https://github.com/tensorflow/tflite-micro/blob/16a9b1e8bfd69a73fcadff210aecf63085d9cbd4/tensorflow/lite/micro/tools/make/Makefile#L404-L408

If you can figure out how to get this working on Windows, that might get you unblocked.

embeddetech, 2021-06-14:
Thank you... so this makefile will not be tested to support Windows? If I work through it perhaps it could be a PR?

Just knowing that I'm not doing something else wrong and that I need to debug this Makefile is very helpful.

advaitjain, 2021-06-14:
Correct, the makefile is not tested on Windows.

But any fixes that you have can certainly be reviewed and merged so please feel free to make a pull request.

embeddetech, 2021-07-06:
Hello @advaitjain,
I was able to get back to this task and work through the issues keeping everything from working on Windows. I have added four python scripts to the make/ folder, and modified for makefile or bash scripts to fix the issues, I hope incorporating a few python scripts to work around the problem of running "Find" on Windows is OK.

If these changes could be pulled in, I would then be able to run the project generator, then from there automate adding the TFLM source code to a Visual Studio C/C++ project. How do you recommend I submit these changes for review? A normal pull request?

Note: I don't see any ..\google Make_internal.inc files, so I wasn't able to fully test that.

embeddetech, 2021-07-08:
Note: In addition, the list_third_party_sources target in the Makefile doesn't appear to correctly print the third party source files. The  list_third_party_headers target seems to work properly. When you run the create_tflm_tree.py script, the third party source code doesn't get copied over as a result of this issue.

petewarden, 2021-07-26:
Just following up on this - Do you have a pull request with your changes? I'd be happy to take a look.

embeddetech, 2021-07-26:
Hi Pete!
I believe I attempted a pull request a few weeks ago but wasn't able to, I'll check again. I have a tool I've written that pulls everything down from GitHub, adds python scripts, modifies makefiles, runs the make file, makes modifications to third party libraries, adds everything to a Visual Studio project, and everything builds and runs perfectly with no errors or warnings. I've slept several times since a few weeks ago when I did that, I need to make sure a pull request will cover everything.

embeddetech, 2021-08-04:
@petewarden FYI I have signed a CLA, but it appears my "Create Pull Request" button is still disabled. Do I just need to wait a little while to see that enabled?

embeddetech, 2021-08-18:
Hi @petewarden and @advaitjain ,
Below is an update on my progress, in case you are interested. Everything is pulled into Visual Studio and Virtuoso and building properly. I'm only appearing to see the "yes" get detected with the speech demo data:

https://youtu.be/Hb0-PUMRyV8

I'm very close to having everything up and running, real system audio should be streaming correctly, I'm just not sure if there's an issue with the LatestAudioTimestamp() that I'm using. If the mock audio worked, then I think I should be able to get the live audio stream working, but at the moment the mock audio isn't quite working either...

Are you still interested in a pull request from me to pull in the changes to let TFLM compile in Visual Studio without errors/warnings?

Thanks!

embeddetech, 2021-08-18:
@petewarden I'm also wondering if the micro speech demo isn't finished yet? I see a note about needing to fix a big with the start_ms being less than zero. Also, it appears that inference is only performed every 100ms when using the mock data, there are 5 new slices (20ms each) added each pass through loop(), but then inference only happens once per pass through loop(), so the LSTM might be expecting to be evaluated on a 20ms window stride, but actually evaluating every 100ms?

advaitjain, 2021-08-19:
> 
> Are you still interested in a pull request from me to pull in the changes to let TFLM compile in Visual Studio without errors/warnings?
> 

Assuming that the change fit well with the existing build system, we would be open to a PR that adds support for Visual Studio.

Also w.r.t. https://github.com/tensorflow/tflite-micro/issues/407#issuecomment-901469738, we can figure out what a reasonable path forward would be.

advaitjain, 2021-09-27:
I'm going to close this bug since there hasn't been any activity in over a month.

closed 2021-09-27 by advaitjain

JavaldeUB, 2023-05-28:
hi,
I'm getting similar issues here.
I have WSL in my windows OS and already clone the repo from tflite-micro. I have everything in the Ubuntu distribution and installed dependencies asked (numpy and Pillow). However, whenever in tfllite-micro calling the Makefile I get this:

sudo make -f tensorflow/lite/micro/tools/make/Makefile test_hello_world_test

tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/kissfft already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
make: *** No rule to make target 'test_hello_world_test'.  Stop.

hello_world folder exists in examples but inside there are evaluate files (.py evaluate_test.cc etc)
I'm getting a bit crazy about this and a hand would be appreciated.
All the best,
Javier