tensorflow/tflite-micro #197: Some problems when I am using the vela compiler (Input(s) and Output tensors must not be dynamic)

dexsr415, 2021-06-17:
@tensorflow/micro

**System information**
- Host OS Platform and Distribution (e.g., Linux Ubuntu 16.04):Windows 10
- TensorFlow installed from (source or binary):anaconda installation
- Tensorflow version (commit SHA if source):2.3.0
- Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.):ethos-U55(I expect to use this)

**Describe the problem**

I want to use ethos-U55, so I need to use the vela compiler.

But the following warning occurred when I used

```
Warning: PACK 'sequential/reshape/Reshape/shape' is not supported on the NPU. Placing on CPU instead
 - Input(s) and Output tensors must not be dynamic
   Op has dynamic tensor(s): sequential/reshape/strided_slice2
Warning: STRIDED_SLICE 'sequential/reshape/strided_slice2' is not supported on the NPU. Placing on CPU instead
 - Input(s) and Output tensors must not be dynamic
   Op has dynamic tensor(s): sequential/reshape/strided_slice2
```

**Please provide the exact sequence of commands/steps when you ran into the problem**

First of all, I'm not sure if it should be posted in this category.

If there is an error, please forgive me.

This is the warning I encountered when using vela compiler

The OP in PACK and STRIDED_SLICE has the warning "Input(s) and Output tensors must not be dynamic"

I'm not sure if this is a coding problem, or if this op will generate these kinds of tensors.

If it is a coding problem, how should I change it?

tflite model:
https://drive.google.com/drive/folders/1x7wA4G2qr4m1wmAxnguCh2uQSGYtl-KL?usp=sharing

The following is the code of my model:
Refer to this website https://www.tensorflow.org/lite/performance/post_training_integer_quant

```
model = tf.keras.Sequential([
        tf.keras.layers.InputLayer(input_shape=(28, 28)),
        tf.keras.layers.Reshape(target_shape=(28,28,1),input_shape=(28,28,1)),
        tf.keras.layers.Conv2D(filters=12, kernel_size=(3,3),activation='relu'),
        tf.keras.layers.MaxPool2D(pool_size=(2, 2)),
        tf.keras.layers.Flatten(),
        tf.keras.layers.Dense(10)
])
```

advaitjain, 2021-06-17:
Since your goal is to use the ethos-u55, I would recommend starting with one of the TfLite Micro examples, such as the [hello_world](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/hello_world/README.md) example.

I'm going to close the current issue, but please feel free to open a new one in the tflite-micro repository for any issues that you run into with the hello_world example.

Also, tagging @freddan80 in case there are any additional ethos-u55 specific examples that might be more relevant here.

closed 2021-06-17 by advaitjain

freddan80, 2021-06-18:
Thx for looping me in. We will check it out. Tagging @jenselofsson and @mansnils .

mansnils, 2021-06-21:
@dexsr415  - As @advaitjain suggested, if you want to get started with Ethos-U55 and Vela in Tflite-micro you could have a look here:
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/cortex_m_corstone_300
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/network_tester
https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/kernels/ethos_u

Running the network example will download Vela and convert the person detect int8 model.

In case you run into specific problems with Vela (outside the scope of the network example and benchmarks where Vela is used in Tflite-micro), you could raise an issue in https://community.arm.com/developer/tools-software/oss-platforms/f/machine-learning-forum/.
You can also use support-ml <support-ml@arm.com> for your issues if you are Arm License customer.

dexsr415, 2021-06-22:
Thank you for your reply

I will check the information you gave.

Thank you so much

dexsr415, 2021-06-22:
> @dexsr415 - As @advaitjain suggested, if you want to get started with Ethos-U55 and Vela in Tflite-micro you could have a look here:
> https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/cortex_m_corstone_300
> https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/network_tester
> https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/kernels/ethos_u
> 
> Running the network example will download Vela and convert the person detect int8 model.
> 
> In case you run into specific problems with Vela (outside the scope of the network example and benchmarks where Vela is used in Tflite-micro), you could raise an issue in https://community.arm.com/developer/tools-software/oss-platforms/f/machine-learning-forum/.
> You can also use support-ml [support-ml@arm.com](mailto:support-ml@arm.com) for your issues if you are Arm License customer.

Thank you for your reply

I will check the information you gave.

Thank you so much

dexsr415, 2021-06-22:
I found that my question is similar to https://github.com/tensorflow/tensorflow/issues/45090#issuecomment-732389188

This problem seems to be related to dynamic size in keras.