tensorflow/tflite-micro #199: Building microlite fails due to missing include fixedpoint.h

jenselofsson, 2021-06-18:
@tensorflow/micro

**System information**
- Host OS Platform and Distribution: Ubuntu 18.04
- TensorFlow installed from (source or binary): source
- Tensorflow version (commit SHA if source): aaa88b7db6cf359d7ceea219a180c6f83775a368
- Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.): x86

**Describe the problem**
Running
```$ make -f tensorflow/lite/micro/tools/make/Makefile microlite```

results in the error:
```
In file included from tensorflow/lite/micro/kernels/activations.cc:18:0:
./tensorflow/lite/kernels/internal/common.h:26:10: fatal error: fixedpoint/fixedpoint.h: No such file or directory
 #include "fixedpoint/fixedpoint.h"
```

The issue seems to be that ```gemmlowp```, which contains fixedpoint.h, isn't being downloaded:
```
$ ls tensorflow/lite/micro/tools/make/downloads/
flatbuffers/
person_model_int8/
pigweed/
```

A workaround to this issue is to run this command before trying to build microlite:
```
$ make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads
$ ls tensorflow/lite/micro/tools/make/downloads/
flatbuffers/
gemmlowp/
kissfft/
person_model_grayscale/
person_model_int8/
pigweed/
ruy/
```

**Please provide the exact sequence of commands/steps when you ran into the problem**
```
$ git clone https://github.com/tensorflow/tflite-micro.git
$ cd tflite-micro/
$ make -f tensorflow/lite/micro/tools/make/Makefile microlite
```

advaitjain, 2021-06-18:
Right, this is a known issue that we haven't gotten around to fixing.

The workaround is exactly what we have in our CI scripts as well:
https://github.com/tensorflow/tflite-micro/blob/56e785e78810291625697dcdaffc66c4dbe32a44/tensorflow/lite/micro/tools/ci_build/test_x86.sh#L29-L30

The fix is to change the gemmlowp download to use a bash script instead of the Makefile macros, similar to https://github.com/tensorflow/tflite-micro/blob/56e785e78810291625697dcdaffc66c4dbe32a44/tensorflow/lite/micro/tools/make/Makefile#L522-L526

advaitjain, 2021-07-22:
Internal bug: http://b/194418406

closed 2021-08-04 by njeffrie

JasmineKaur-eaton, 2023-12-25:
Hi, I am still getting the same error and unable to find the above files to execute cmd.
How to fix it?
I have downloaded the tfite-micro_main from GIT. 
any help would be appreciated.

thorinf, 2024-09-17:
Is this issue resovled? I'm still having issues