tensorflow/tflite-micro #201: MicroProfiler difficult to subclass

alanvgreen, 2021-06-18:
I would like to make a subclass of [tflite::MicroProfiler](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_profiler.h#L29) which formats data differently. In particular, I would like CSV output to make it easy to use performance data in a spreadsheet. 

However, since the data members of MicroProfiler are all private, and there are no accessors, it is difficult to build this as a subclass.

Here are several alternatives:

1. Make the data members protected instead of private.
2. Provide accessor methods so that the data collected by the profiler can be used by subclasses (and possibly by other classes too)
3. Split Split MicroProfiler into an interface and a separate implementation. 
4. Make a custom version of MicroProfiler.

What is the best option here? I'm happy to send a PR if that would be helpful.

advaitjain, 2021-06-18:
Having CSV output sounds generally useful so for this particular case I would say

  5. Add CSV output to tflite::MicroProfiler.

I'm not too concerned about the size increase since the ScopedMicroProfiler is a noop builds where we strip all the strings to reduce the size https://github.com/tensorflow/tflite-micro/blob/56e785e78810291625697dcdaffc66c4dbe32a44/tensorflow/lite/micro/micro_graph.cc#L164-L171

I can probably go with either 2, 3, or 5. And I'm possibly leaning towards 5 since csv output could be generally useful.

What do you think?

alanvgreen, 2021-06-19:
5 sounds good to me too. One less thing to maintain downstream :)

I'll send a change.

petewarden, 2021-07-26:
Closing this, since the pull request has been merged.

closed 2021-07-26 by petewarden