tensorflow/tflite-micro #202: Build TFLite Micro for riscv32_mcu

chuantunglin, 2021-06-19:
**System information**

- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux Ubuntu 20.04
- TensorFlow installed from (source or binary): N/A
- TensorFlow version: Tensorflow 2.4.2
- Python version: 3.8.5
- Installed using virtualenv? pip? conda?: N/A
- Bazel version (if compiling from source): N/A
- GCC/Compiler version (if compiling from source): gcc 9.3.0, riscv64-unknown-elf-gcc (GCC) 10.2.0
- CUDA/cuDNN version: N/A
- GPU model and memory: N/A
- Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.): RISC-V

**Describe the problem**
Cannot build TFLite Micro for RISC-V target (TARGET=riscv32_mcu). The error message is shown at the bottom.
The command below doesn't work since the file riscv32_mcu_makefile.inc doesn't exist.
_make -f tensorflow/lite/micro/tools/make/Makefile TARGET=riscv32_mcu test_hello_world_test_

Does anyone successfully build the hello_world example for RISCV (riscv32_mcu)?
What are the exact commands and steps?

**Provide the exact sequence of commands / steps that you executed before running into the problem**

1. git clone --depth 1 https://github.com/tensorflow/tensorflow.git
2. cd tensorflow
3. vi tensorflow/lite/micro/tools/make/targets/mcu_riscv_makefile.inc
4. change _ifeq ($(TARGET), riscv32_mcu)_ to _ifeq ($(TARGET), mcu_riscv)_
5. _make -f tensorflow/lite/micro/tools/make/Makefile TARGET=riscv32_mcu hello_world_

**Any other info / logs**
Include any logs or source code that would be helpful to diagnose the problem. If including tracebacks, please include the full traceback. Large logs and files should be attached.

$ make -f tensorflow/lite/micro/tools/make/Makefile TARGET=mcu_riscv TARGET_ARCH=riscv32_mcu test_hello_world_test
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/person_model_int8 already exists, skipping the download.
riscv64-unknown-elf-g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wmissing-field-initializers -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wstrict-aliasing -Wno-unused-parameter -DMCU_RISCV -march=rv32imac -mabi=ilp32 -mcmodel=medany -mexplicit-relocs -fno-builtin-printf -fno-exceptions -DTF_LITE_MCU_DEBUG_LOG -DTF_LITE_USE_GLOBAL_CMATH_FUNCTIONS -fno-unwind-tables -ffunction-sections -fdata-sections -funsigned-char -Wvla -Wall -Wextra -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wmissing-field-initializers -Wno-unused-parameter -Wno-write-strings -Wunused-function -fno-delete-null-pointer-checks -fno-threadsafe-statics -fomit-frame-pointer -fno-use-cxa-atexit -Os -fpermissive -fno-rtti --std=gnu++11 -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/include -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/drivers/ -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env/freedom-e300-hifive1 -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite/micro/riscv32_mcu/debug_log.cc -o tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/riscv32_mcu/debug_log.o
riscv64-unknown-elf-g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wmissing-field-initializers -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wstrict-aliasing -Wno-unused-parameter -DMCU_RISCV -march=rv32imac -mabi=ilp32 -mcmodel=medany -mexplicit-relocs -fno-builtin-printf -fno-exceptions -DTF_LITE_MCU_DEBUG_LOG -DTF_LITE_USE_GLOBAL_CMATH_FUNCTIONS -fno-unwind-tables -ffunction-sections -fdata-sections -funsigned-char -Wvla -Wall -Wextra -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wmissing-field-initializers -Wno-unused-parameter -Wno-write-strings -Wunused-function -fno-delete-null-pointer-checks -fno-threadsafe-statics -fomit-frame-pointer -fno-use-cxa-atexit -Os -fpermissive -fno-rtti --std=gnu++11 -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/include -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/drivers/ -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env/freedom-e300-hifive1 -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite/micro/kernels/activations.cc -o tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/kernels/activations.o
In file included from tensorflow/lite/micro/kernels/activations.cc:18:
./tensorflow/lite/kernels/internal/common.h:26:10: fatal error: fixedpoint/fixedpoint.h: No such file or directory
   26 | #include "fixedpoint/fixedpoint.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [tensorflow/lite/micro/tools/make/Makefile:649: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/kernels/activations.o] Error 1

advaitjain, 2021-06-19:
Please try the workaround from https://github.com/tensorflow/tflite-micro/issues/199:

```
make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads
```

I'm going to close this issue as a duplicate of #199.

closed 2021-06-19 by advaitjain

chuantunglin, 2021-06-19:
> Please try the workaround from #199:
> 
> ```
> make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads
> ```
> 
> I'm going to close this issue as a duplicate of #199.

The error is solved but I got another error: #203