tensorflow/tflite-micro #203: Build TFLite Micro for riscv32_mcu

chuantunglin, 2021-06-19:
This issue is different from #202

**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux Ubuntu 20.04
- TensorFlow installed from (source or binary): N/A
- TensorFlow version: Tensorflow 2.4.2
- Python version: 3.8.5
- Installed using virtualenv? pip? conda?: N/A
- Bazel version (if compiling from source): N/A
- GCC/Compiler version (if compiling from source): gcc 9.3.0, riscv64-unknown-elf-gcc (GCC) 10.2.0
- CUDA/cuDNN version: N/A
- GPU model and memory: N/A
- Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.): RISC-V

**Describe the problem**
Cannot build TFLite Micro for RISC-V target (TARGET=riscv32_mcu). The error message is shown at the bottom.
The command below doesn't work since the file _riscv32_mcu_makefile.inc_ doesn't exist.
`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=riscv32_mcu hello_world`

Does anyone successfully build the hello_world example for RISCV (riscv32_mcu)?
What are the exact commands and steps?

**Provide the exact sequence of commands / steps that you executed before running into the problem**
1. `git clone https://github.com/tensorflow/tflite-micro.git`
2. `cd tflite-micro/`
3. `make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads`
4. `vi tensorflow/lite/micro/tools/make/targets/mcu_riscv_makefile.inc`
5. change `ifeq ($(TARGET), riscv32_mcu)` to `ifeq ($(TARGET), mcu_riscv)`
6. `make -f tensorflow/lite/micro/tools/make/Makefile PARSE_THIRD_PARTY=true TARGET=mcu_riscv TARGET_ARCH=riscv32_mcu generate_hello_world_make_project`
7. `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=mcu_riscv TARGET_ARCH=riscv32_mcu hello_world`

**Any other info / logs**
Include any logs or source code that would be helpful to diagnose the problem. If including tracebacks, please include the full traceback. Large logs and files should be attached.

$ make -f tensorflow/lite/micro/tools/make/Makefile TARGET=mcu_riscv TARGET_ARCH=riscv32_mcu hello_world               tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/person_model_int8 already exists, skipping the download.
riscv64-unknown-elf-g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wmissing-field-initializers -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wstrict-aliasing -Wno-unused-parameter -DMCU_RISCV -march=rv32imac -mabi=ilp32 -mcmodel=medany -mexplicit-relocs -fno-builtin-printf -fno-exceptions -DTF_LITE_MCU_DEBUG_LOG -DTF_LITE_USE_GLOBAL_CMATH_FUNCTIONS -fno-unwind-tables -ffunction-sections -fdata-sections -funsigned-char -Wvla -Wall -Wextra -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wmissing-field-initializers -Wno-unused-parameter -Wno-write-strings -Wunused-function -fno-delete-null-pointer-checks -fno-threadsafe-statics -fomit-frame-pointer -fno-use-cxa-atexit -Os -fpermissive -fno-rtti --std=gnu++11 -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/include -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/drivers/ -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env -Itensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env/freedom-e300-hifive1 -Itensorflow/lite/micro/tools/make/downloads/kissfft -o tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/hello_world tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/examples/hello_world/main.o tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/examples/hello_world/main_functions.o tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/examples/hello_world/model.o tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/examples/hello_world/output_handler.o tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/examples/hello_world/constants.o tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/lib/libtensorflow-microlite.a -Wl,--fatal-warnings -Wl,--gc-sections -Ttensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env/freedom-e300-hifive1/flash.lds -nostartfiles -Ltensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env --specs=nano.specs -lm
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/examples/hello_world/main.o: Relocations in generic ELF (EM: 62)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/examples/hello_world/main.o: Relocations in generic ELF (EM: 62)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/obj/core/tensorflow/lite/micro/examples/hello_world/main.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make: *** [tensorflow/lite/micro/examples/hello_world/Makefile.inc:44: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/hello_world] Error 1

petewarden, 2021-07-26:
It looks like the key problem was the -nostartfiles argument being passed to the linker. I'm not sure what the history was, but I've removed that in PR #321. I've filed #322 to track automated testing of this platform.

petewarden, 2021-07-27:
Now that #321 is checked in, you should be able to use that as a workflow to build for RISC-V, at least for that particular board. I'm closing this bug, unless you have extra information?

closed 2021-07-27 by petewarden