tensorflow/tflite-micro #213: keil projects build unexpected big executable

RockySong, 2021-06-23:
By some investigation, keil projects build will link about 83KB unwanted C++ library objects, including the largest locale.cpp.o (~55KB).
I've already found the inclusion of below files will cause keil (ARM AC6) to link them:
<iomanip>
<complex>

Which are included or indirectly included by kernel_utils.cc.

It seems <iomanip> is not required, but <complex> is required for succesful build.

advaitjain, 2021-06-23:
While the TFLM library does contain many unused symbols, when linking the final executable, all unused symbols will be dropped by the linker. We observe this behavior with the various tests, examples and benchmark binaries in the TFLM repository when using arm-gcc and various other toolchains (e.g. Xtensa and Hexagon).

It appears from your description that that is not what you are observing with your Keil project. Tagging @freddan80 to answer this specifically for Keil projects.

RockySong, 2021-06-24:
Thanks Advait, that's true.
From further investigation, I found kernel_util.cc indirectly included
<iomanip>
<complex>
If these two headers are included, keil will link extra about 83KB (AC 6.14 in KEIL 5.34) or even 160KB (AC 6.12 in KEIL 5.27) C++ libraries.
The keil projects are generated by the tflm makefile, I used the helloworld example, and the size information is from the helloworld.map generated by keil..

freddan80, 2021-06-28:
@RockySong you're correct. We observed this issue about a month back, and there is a fix that will be a part of the upcoming release of Arm compiler 6, most likely in August. Please check [this](https://developer.arm.com/tools-and-software/embedded/arm-compiler/downloads/version-6) page for updates. Sorry for the inconvenience.

freddan80, 2021-06-28:
Tagging @MatthiasHertel80

RockySong, 2021-07-03:
Noted with thanks!

closed 2021-07-03 by RockySong