tensorflow/tflite-micro #242: The kernel sources are compiled into both MICROLITE_LIB_OBJS and MICROLITE_KERNELS_OBJS

jenselofsson, 2021-07-01:
**System information**
- Host OS Platform and Distribution: Ubuntu 18.04
- TensorFlow installed from: source
- Tensorflow version (commit SHA if source): e172d86

**Describe the problem**
We've noticed that the kernel sources seems to be compiled into both the MICROLITE_LIB_OBJS and MICROLITE_KERNEL_OBJS. This leads to the kernels sources being compiled with both CORE_OPTIMIZATION_LEVEL (-Os) and KERNEL_OPTIMIZATION_LEVEL (-O2) depending on which object file they're found in.

This might lead to the final binary using kernel source being built with -Os, instead of -O2 which means that the execution of the kernels will be slower. armclang will also display a warning saying `L6876W: Minor variants of archive member '<member>' include multiple base variants`

Removing line 612 from [tensorflow/lite/micro/tools/make/Makefile](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/make/Makefile#L612) seems to solve the issue at first glance.

**Please provide the exact sequence of commands/steps when you ran into the problem**
`$ make -f tensorflow/lite/micro/tools/make/Makefile TARGET=cortex_m_corstone_300 TARGET_ARCH=cortex-m55 BUILD_TYPE=release_with_logs TOOLCHAIN=armclang test_hello_world_test`

jenselofsson, 2021-07-06:
The problem seems to be a bit more complex than the [proposed solution](https://github.com/jenselofsson/tflite-micro/commit/36f13902052d5c7b6a8ac8503301aee2b2f3598c). If I run the script tensorflow/lite/micro/tools/ci_build/test_all.sh, the Arduino build fails with the following errors:
```
/tmp/tflite-arduino-build/libraries/tensorflow_lite/tensorflow/lite/micro/all_ops_resolver.cpp.o: In function `tflite::AllOpsResolver::AllOpsResolver()':
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:118: undefined reference to `tflite::ops::micro::Register_ABS()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:123: undefined reference to `tflite::ops::micro::Register_ADD()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:128: undefined reference to `tflite::Register_ADD_N()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:134: undefined reference to `tflite::ops::micro::Register_ARG_MAX()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:139: undefined reference to `tflite::ops::micro::Register_ARG_MIN()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:144: undefined reference to `tflite::Register_AVERAGE_POOL_2D()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:149: undefined reference to `tflite::Register_BATCH_TO_SPACE_ND()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:157: undefined reference to `tflite::ops::micro::Register_CEIL()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:168: undefined reference to `tflite::ops::micro::Register_CONCATENATION()'
/tmp/tflite-arduino-build/libraries/tensorflow_lite/tensorflow/lite/micro/all_ops_resolver.cpp.o: In function `tflite::AllOpsResolver::AllOpsResolver()':
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:173: undefined reference to `tflite::Register_CONV_2D()'
/tmp/tflite-arduino-build/libraries/tensorflow_lite/tensorflow/lite/micro/all_ops_resolver.cpp.o: In function `tflite::AllOpsResolver::AllOpsResolver()':
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:178: undefined reference to `tflite::ops::micro::Register_COS()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:183: undefined reference to `tflite::Register_CUMSUM()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:189: undefined reference to `tflite::Register_DEPTH_TO_SPACE()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:194: undefined reference to `tflite::Register_DEPTHWISE_CONV_2D()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:199: undefined reference to `tflite::ops::micro::Register_DEQUANTIZE()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:204: undefined reference to `tflite::Register_DETECTION_POSTPROCESS()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:209: undefined reference to `tflite::Register_ELU()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:214: undefined reference to `tflite::ops::micro::Register_EQUAL()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:218: undefined reference to `tflite::Register_ETHOSU()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:220: undefined reference to `tflite::GetString_ETHOSU()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:240: undefined reference to `tflite::ops::micro::Register_FLOOR()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:244: undefined reference to `tflite::Register_FLOOR_DIV()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:249: undefined reference to `tflite::Register_FLOOR_MOD()'
/tmp/tflite-arduino-build/libraries/tensorflow_lite/tensorflow/lite/micro/all_ops_resolver.cpp.o: In function `tflite::AllOpsResolver::AllOpsResolver()':
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:254: undefined reference to `tflite::Register_FULLY_CONNECTED()'
/tmp/tflite-arduino-build/libraries/tensorflow_lite/tensorflow/lite/micro/all_ops_resolver.cpp.o: In function `tflite::AllOpsResolver::AllOpsResolver()':
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:271: undefined reference to `tflite::ops::micro::Register_GREATER()'
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:276: undefined reference to `tflite::ops::micro::Register_GREATER_EQUAL()'
[... cut ...]
/home/username/Arduino/libraries/tensorflow_lite/src/tensorflow/lite/micro/micro_mutable_op_resolver.h:527: undefined reference to `tflite::ops::micro::Register_UNPACK()'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "TensorFlowLite.h"
 Used: /home/username/Arduino/libraries/tensorflow_lite
Error: build failed: exit status 1
Error during build
Cleaning up to prevent Kokoro errors (see http://b/186570469)
find: ‘../google/’: No such file or directory
rm -rf tensorflow/lite/micro/tools/make/gen
rm -rf tensorflow/lite/micro/tools/make/downloads
```

advaitjain, 2021-07-07:
Thanks for creating the issue. We'll look into this.

njeffrie, 2021-07-07:
Looks like an error on my part during the Makefile source refactor. Your change plus adding MICROLITE_CC_KERNEL_SRCS to ALL_SRCS (Makefile line 614) appears to fix the issue.

jenselofsson, 2021-07-08:
The issue seems to persist even with those [changes](https://github.com/jenselofsson/tflite-micro/commit/d054a359d7220839c259ea9bb231192fcf79991c) (if I ran the github actions correctly, it's the first time I tried):
https://github.com/jenselofsson/tflite-micro/runs/3018202702?check_suite_focus=true

I tried looking around in the test scripts and in the Makefile to see if I could spot if anything is needed particularly related to Arduino, but i couldn't find anything obvious (although I'm not too experienced with how the Arduino build system works).

closed 2021-07-08 by advaitjain