tensorflow/tflite-micro #246: Broken project generation for ARC

gerbauz, 2021-07-02:
**System information**
- Host OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Any
- TensorFlow installed from (source or binary): Source
- Tensorflow version (commit SHA if source): Latest from main branch [tflite-micro](https://github.com/tensorflow/tflite-micro)
- Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.): ARC

Hi, I've noticed some changes in micro Makefile, which caused troubles in project generation for ARC platform.
These lines: [link](https://github.com/tensorflow/tflite-micro/blob/61952b5020877d8426f28684cbe511b178a60054/tensorflow/lite/micro/tools/make/Makefile#L241-L247) affects the name of the folder, where the project will be generated. 
Sometime ago, these lines were here: [link](https://github.com/tensorflow/tflite-micro/blob/3e37fdaabacdc198d26608a6f4ceb3b809751097/tensorflow/lite/micro/tools/make/Makefile#L607-L613) under include of target specific makefile: [link](https://github.com/tensorflow/tflite-micro/blob/3e37fdaabacdc198d26608a6f4ceb3b809751097/tensorflow/lite/micro/tools/make/Makefile#L555). So, this change just broke up project generation for ARC, as we used TARGET value to add the name of tcf file which was used to build embARC MLI library and will be used to run application on nSIM.

So, can I fix it moving these lines back where they were (under include of specific make file)?

advaitjain, 2021-07-07:
For an quick fix, making changes in the target-specific makefile works.

Please note that we would like to completely remove project generation from the Malefiles and have all project generation instead use [this Python script](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py).

Ideally we would like to remove all Makefile-based project generation by the end of 2021.

We expect to have an example of using the Python project generation with the Arduino in the next month or two and then start asking TFLM users to do the same for their targets.

gerbauz, 2021-08-06:
Hi @advaitjain, I just posted a fix in some of make files and while testing I noticed, that third party downloads are no longer downloaded, when I try to generate microlite targed.

Otherwise, when I change this:

```
GENDIR := $(MAKEFILE_DIR)/gen/$(TARGET)_$(TARGET_ARCH)_$(BUILD_TYPE)/
CORE_OBJDIR := $(GENDIR)obj/core/
KERNEL_OBJDIR := $(GENDIR)obj/kernels/
THIRD_PARTY_OBJDIR := $(GENDIR)obj/third_party/
BINDIR := $(GENDIR)bin/
LIBDIR := $(GENDIR)lib/
PRJDIR := $(GENDIR)prj/
```

To this:
```
GENDIR := $(MAKEFILE_DIR)/gen/$(TARGET)_$(TARGET_ARCH)_$(BUILD_TYPE)/
OBJDIR := $(GENDIR)obj/
BINDIR := $(GENDIR)bin/
LIBDIR := $(GENDIR)lib/
PRJDIR := $(GENDIR)prj/
```
Script successfully starts third party downloads. You can find that lines [here](https://github.com/tensorflow/tflite-micro/blob/abb5e1546695373d8e1d2a34ae14747dd4b446d3/tensorflow/lite/micro/tools/make/Makefile#L614-L621).

What do you think can cause this problem?

advaitjain, 2021-08-06:
I'm unsure about the cause of the third party downloads not occurring. What I will say is that downloading via Makefile macros has historically been error-prone and hard to maintain.

Instead we are moving towards having the downloads be via a script.

https://github.com/tensorflow/tflite-micro/blob/6b8b699ea8d9f041089016bb5a3081ff20d77315/tensorflow/lite/micro/tools/make/Makefile#L522-L526

https://github.com/tensorflow/tflite-micro/blob/6b8b699ea8d9f041089016bb5a3081ff20d77315/tensorflow/lite/micro/tools/make/ext_libs/xtensa.inc#L12-L15

My suggestion would be to change your downloads to a bash script.

closed 2021-08-10 by gerbauz