tensorflow/tflite-micro #263: Register_SOFTMAX_INT8_INT16 redefined error

anilmaddala, 2021-07-09:
Following [https://codelabs.developers.google.com/codelabs/sparkfun-tensorflow/#3](url) to build micro speech example.

After executing `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge micro_speech_bin`

Running into 
`tensorflow/lite//micro/kernels/xtensa/softmax_int8_int16.cc:285:20: error: redefinition of 'TfLiteRegistration tflite::Register_SOFTMAX_INT8_INT16()'
  285 | TfLiteRegistration Register_SOFTMAX_INT8_INT16() {
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from tensorflow/lite//micro/kernels/xtensa/softmax_int8_int16.cc:25:
./tensorflow/lite/micro/kernels/softmax.h:38:27: note: 'TfLiteRegistration tflite::Register_SOFTMAX_INT8_INT16()' previously defined here
   38 | inline TfLiteRegistration Register_SOFTMAX_INT8_INT16() {
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [tensorflow/lite/micro/tools/make/Makefile:654: tensorflow/lite/micro/tools/make/gen/sparkfun_edge_x86_64_default/obj/core/tensorflow/lite//micro/kernels/xtensa/softmax_int8_int16.o] Error 1`

advaitjain, 2021-07-09:
The sparkfun_edge continuous build is green: https://github.com/tensorflow/tflite-micro#community-supported-builds

And I just tested the command locally and the error is not reproducible.

The fact that your command is attempting to build a file from kernels/xtensa for `TARGET=sparkfun_edge` (tensorflow/lite//micro/kernels/xtensa/softmax_int8_int16.cc) suggests that there may be something wrong with your setup.

kernels/xtensa will only get pulled u=in if you specify `OPTIMIZED_KERNEL_DIR=xtensa` as part of your make command.

closed 2021-07-09 by advaitjain

advaitjain, 2021-07-20:
Here is additional context from a different issue describing the same error: https://github.com/tensorflow/tflite-micro/issues/295#issuecomment-883139053