tensorflow/tflite-micro #280: quantize.cpp:58 input->type == kTfLiteFloat32 || input->type == kTfLiteInt16 || inp was not true

AdamMiltonBarker, 2021-07-14:
**System information**
- Host OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Ubuntu 18.04
- TensorFlow installed from (source or binary): binary
- Tensorflow version (commit SHA if source): 2.5.0
- Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.): Arduino Nano 33 BLE Sense

**Describe the problem**

I have converted our CNN to TFLite using full integer quantization (in and out: uint8)
AllocateTensors fails with quantize.cpp:58 input->type == kTfLiteFloat32 || input->type == kTfLiteInt16 || inp was not true

[image]

advaitjain, 2021-07-14:
TFLM no longer supports the deprecated uint8 quantization (https://github.com/tensorflow/tflite-micro/issues/216). Our recommendation is to use int8 quantization.

closed 2021-07-14 by advaitjain

AdamMiltonBarker, 2021-07-14:
Ok your documentation really needs updating, all Github links are dead and there is no mention of this in this in the docs. Thanks for letting me know.