tensorflow/tflite-micro #282: Not able to call Invoke() in the person-detection model

stanleycelestin1, 2021-07-14:
I have taken the person-detection model and tried to configure it for the stm32f746 discover board. I have not been able to successfully call invoke and I get this on my serial bus:

```
Type UINT8 (3) not supported.
Node DEPTHWISE_CONV_2D (number 0) failed to invoke with status 1
```

Can anyone help me understand what I may be doing wrong?

Here's a code snippet:
`printf("\n\n\n");
  printf("Start of Main  . . . . . . . \n");
  tflite::ErrorReporter* error_reporter = nullptr;
  const tflite::Model* model = nullptr;

  tflite::MicroInterpreter* interpreter = nullptr;
  TfLiteTensor* input = nullptr;
  // An area of memory to use for input, output, and intermediate arrays.
  constexpr int kTensorArenaSize = 136 * 1024;
  static uint8_t tensor_arena[kTensorArenaSize];

   // begin
  static tflite::MicroErrorReporter micro_error_reporter;
  error_reporter = &micro_error_reporter;

  // GET MODEL
  model = tflite::GetModel(g_person_detect_model_data);
  if (model->version() != TFLITE_SCHEMA_VERSION) {
    TF_LITE_REPORT_ERROR(error_reporter,
                         "Model provided is schema version %d not equal "
                         "to supported version %d.",
                         model->version(), TFLITE_SCHEMA_VERSION);
  } else {
    printf("Succefuly got model.\n");
  }

  // GET needed Operations
  printf("Gettting needed operations.\n");

  static tflite::MicroMutableOpResolver<5> micro_op_resolver;
  // micro_op_resolver.AddAveragePool2D();
  // micro_op_resolver.AddConv2D();
  // micro_op_resolver.AddDepthwiseConv2D();
  // micro_op_resolver.AddReshape();
  // micro_op_resolver.AddSoftmax();

  if (micro_op_resolver.AddAveragePool2D() != kTfLiteOk) {
    printf("Could not get op AddAveragePool2D().\n");
  }
  if (micro_op_resolver.AddConv2D() != kTfLiteOk) {
    printf("Could not get op AddConv2D().\n");
  }
  if (micro_op_resolver.AddDepthwiseConv2D() != kTfLiteOk) {
    printf("Could not get op AddDepthwiseConv2D().\n");
  }
  if (micro_op_resolver.AddReshape() != kTfLiteOk) {
    printf("Could not get op AddReshape().\n");
  }
  if (micro_op_resolver.AddSoftmax() != kTfLiteOk) {
    printf("Could not get op AddSoftmax().\n");
  }
  

  printf("Finished gettting needed operations. Creating static_interpreter ...\n");

  static tflite::MicroInterpreter static_interpreter(
      model, micro_op_resolver, tensor_arena, kTensorArenaSize, error_reporter);
  interpreter = &static_interpreter;

  printf("Created static_interpreter. Allocating Tensors ...\n");

  // // Allocate memory from the tensor_arena for the model's tensors.
  TfLiteStatus allocate_status = interpreter->AllocateTensors();
  printf("Allocated Tensors with status: %d.\n", allocate_status);
  
  ThisThread::sleep_for(500ms);
  input = interpreter->input(0);

  // print input information
  printf("size:           %d\n", input->dims->size);
  printf("data[0]:        %d\n", input->dims->data[0]);
  printf("kNumRows:       %d\n", input->dims->data[1]);
  printf("kNumCols:       %d\n", input->dims->data[2]);
  printf("kNumChannels:   %d\n", input->dims->data[3]);
  printf("kTfLiteInt8 ?:  %d\n", input->type);

  ThisThread::sleep_for(500ms);

  printf("Starting copying the g_person_data to input->data.int8.\n");
  // memcpy(input->data.int8, g_person_data, input->bytes); // g_person is in uint8_t need to convert to int8_t

  // for(int i=0; i<g_person_data_size; i++) {
  //   input->data.int8[i] =  static_cast<int8_t>(g_person_data[i]);
  // }

  ThisThread::sleep_for(1500ms);
  printf("Finished memcpying the g_person_data to input->data.int8. Invoking interpreter . . .\n");
  TfLiteStatus invoke_status = interpreter->Invoke();

  ThisThread::sleep_for(1500ms);
  printf("Invoke() finhsed running.\n");
  if (invoke_status != kTfLiteOk) {
    printf("Invoke() FAILED.\n");
  } else {
    printf("Invoke() SUCCEEDED.\n");
  }`

stanleycelestin1, 2021-07-14:
Hmm I think there is a mismatch:

input->type returns 3 which is uint8 but `Type UINT8 (3) not supported.`

advaitjain, 2021-07-14:
TFLM no longer supports uint8 quatization (https://github.com/tensorflow/tflite-micro/issues/216).

The person detection example https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/person_detection has an int8 model. Can you try using that.

Tagging @petewarden and @freddan80 for STM32 specific issue.

closed 2021-07-19 by stanleycelestin1