tensorflow/tflite-micro #284: Didn't find op for builtin opcode 'SOFTMAX' version '2'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

AdamMiltonBarker, 2021-07-15:
I have a model that is 179960 len, had quite a lot of issues with the arena maxing out memory when it gets to the interpreter. It was requiring 627 x 1024 but that was crashing memory. I now have the arena set at 200 x 1024 and using micro op resolver, and suddenly it started working, then I was hit with this. 

`Didn't find op for builtin opcode 'SOFTMAX' version '2'`

I know it can be solved with AllOpsResolver but then throws the original issue:

`Needed 627488 but only 203072 was available.`

Can anyone suggest how can be solved without having to use AllOpsResolver, haven't been able to find this issue documented anywhere. Thanks in advance.

advaitjain, 2021-07-15:
Can you make sure that all the ops in your model have been explicitly added https://github.com/tensorflow/tflite-micro/blob/2ddb3cc8007ba963b73a2c949050ec6f7593113b/tensorflow/lite/micro/micro_mutable_op_resolver.h#L455

Here's an example from the keyword benchmark 
https://github.com/tensorflow/tflite-micro/blob/2ddb3cc8007ba963b73a2c949050ec6f7593113b/tensorflow/lite/micro/benchmarks/keyword_benchmark.cc#L54-L58

AdamMiltonBarker, 2021-07-15:
Thank you for quick reply. These are the ops:

  static tflite::MicroMutableOpResolver<2> micro_op_resolver;
  micro_op_resolver.AddPad();
  micro_op_resolver.AddConv2D();
  micro_op_resolver.AddPad();
  micro_op_resolver.AddConv2D();
  micro_op_resolver.AddMaxPool2D();
  micro_op_resolver.AddReshape();
  micro_op_resolver.AddFullyConnected();
  micro_op_resolver.AddSoftmax();

I will check out the links you gave, thanks buddy.

AdamMiltonBarker, 2021-07-15:
Ah I found out why it suddenly started working, I reduced the number in MicroMutableOpResolver<8> to MicroMutableOpResolver<5>, if I change it back to 8 it gives me the original erro  Arena size is too small for all buffers. Needed 627488 but only 203072 was available.

So my question is, when my model is almost half the size of the person detection, and the code is also smaller (less files) why does it need 627 x 1024 when the model from person detector only needs 136 ?

AdamMiltonBarker, 2021-07-15:
Ah I remember now the change I made and why  it suddenly started working, I reduced the ops number in MicroMutableOpResolver<8> to MicroMutableOpResolver<5>, if I change it back to 8 it gives me the original error Arena size is too small for all buffers. Needed 627488 but only 203072 was available.

So my question is, when my model is almost half the size of the person detection why does it need 627 x 1024 when the model from person detector only needs 136 ? The code is also smaller (less files) but it maxes out. Is it the additional layers ? Any suggestions on getting this working have been banging my head around for days.

AdamMiltonBarker, 2021-07-15:
Ive found I get away with removing the 2nd Pad and Conv2D, but I am figuring that is only because they have already been provided I think, so I am stuck at a minimum of 6 ops and that gives the same error, Needed 627488 but only 203072 was available. Is the only option here build a network with less layers ?

AdamMiltonBarker, 2021-07-15:
OK I now understand more about ops in tflite and where I made my mistake. I made a new model, this has actually been a blessing in disguise. The model is uploading fine to Arduino, and the model is pretty damn good. Sorry for the noob questions but you helped me understand more about micro models :) 

[image]
[image]
[image]
[image]
[image]
[image]

closed 2021-07-15 by AdamMiltonBarker