tensorflow/tflite-micro #286: Sigmoid,tanh: TFL int16 reference code compared to float implementation is off by 3-4 bits

lukmanr-cadence, 2021-07-15:
@advaitjain , @njeffrie, @petewarden, @nyadla-sys 

### Describe the problem
TFL int16 reference code compared to float implementation is off by 3-4 bits (int8 is off by 1 bit).
TFLM does not have int16 support yet.
Sigmoid shows maxdiff of 6 (3 bit).
Tanh shows maxdiff of 12 (4 bit).
Cadence’s hardware implementation is 1 bit off and that’s how it was discovered that TFLM implementation was off
A bug with test vectors will help the TFLM team figure out what needs to be done here

### Source code / logs
[input_file_sigmoid_Q12.txt](https://github.com/tensorflow/tensorflow/files/6232546/input_file_sigmoid_Q12.txt)
[input_file_tanh_Q12.txt](https://github.com/tensorflow/tensorflow/files/6232547/input_file_tanh_Q12.txt)
Attached couple of files with input data which shows maximum difference. This input list is for Q12 (1 sign bit, 3 integer bits, 12 fractional bits).  We also observed 3-4 bit difference for other Q format as well (Q11 and Q15).

advaitjain, 2021-07-15:
> @petewarden :A Gentle reminder.!

done

dominicsymes, 2021-10-05:
Hi. Can I just ask which code this issue refers to as the TFL int16 reference code.
If it is reference_ops::Tanh() that calls gemlowp::tanh() then this has a difference of 6 for sigmoid and 12 for tanh for int16 Q12 input.
However, I think TFLite no longer uses this for reference for int16 tanh and now uses reference_integer_ops::Tanh() that is more accurate, within I think +/1 for int16 Q12
(for example https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/kernels/activations.cc#L945)
So, I do not think that TFLite int16 has an issue with Tanh/sigmoid precision.

lukmanr-cadence, 2021-11-09:
Hi,
We tested  reference_integer_ops::Logistic() & reference_integer_ops::Tanh() for all possible Q12 inputs.
reference_integer_ops::Logistic() is just 1 bit off for Q12 inputs.
But reference_integer_ops::Tanh() is 2 bit off for Q12 inputs and this difference increases from 2 bit to 4 bit when the input changes from Q12 to Q15.

tingyan19, 2022-01-25:
I dug a little deeper and found this:

https://github.com/tensorflow/tensorflow/blob/ee62dbd6d71243c0d7c79123f6b34d26cc048599/tensorflow/lite/kernels/activations_test.cc#L198-L215

So for Int16, 4 bits of error (2^-12) is actually expected for the TFLite implementation based on trade-off of accuracy and time/memory cost. 

I will try to port the TFLite tests for int16 tanh and sigmoid (logistic) over to TFLM.

closed 2022-01-30 by tingyan19