tensorflow/tflite-micro #287: Strange outputs

AdamMiltonBarker, 2021-07-16:
Arduino 33 BLE Sense
Tensorflow 2.5.0

I have the following model, on Linux this model classifies all 20 images in an Acute Lymphoblastic Leukemia test set correctly:

  micro_op_resolver.AddAveragePool2D();
  micro_op_resolver.AddConv2D();
  micro_op_resolver.AddDepthwiseConv2D();
  micro_op_resolver.AddReshape();
  micro_op_resolver.AddFullyConnected();
  micro_op_resolver.AddSoftmax();

On Arduino 33 BLE Sense I am using JpegDecoder to decode the images and pass them to input.data.int8, I have also tested with the instructions for converting to c array in the following link: https://github.com/tensorflow/tflite-micro/blob/f583f92992c3c9dfb8e10f36d66b2fe7267cf7bc/tensorflow/lite/micro/examples/person_detection/person_image_data.h#L17-L20 with exact same results.

It is not working well :D On Arduino I am always receiving  an integer for positive and the negative of the integer for negative. IE:

Positive image:
Positive score: 111
Negative score: -111

Negative image:
Positive score: 74
Negative score: -74

Nearly always 74, -74 for negative and nearly always 111, -111 for positive.

At first I was simply reading  the image in using SD.open ........ jpegFile.read() and looping through assigning to input.data.int8, but then I made a version that uses decodeSdFile and made a script based on the person detector project that does the grayscaling etc, but either way gets exactly the same results as above. 

Do you have any suggestions as to why this is happening, I don't have the experience with Arduino ML to work this out. Thanks in advance.

AdamMiltonBarker, 2021-07-16:
This is nuts. I have the same image being processed which is a negative sample. In loop I have:

```
  if (kTfLiteOk != getImage(images[6], input->data.int8)) {
    TF_LITE_REPORT_ERROR(error_reporter, "Image capture failed.");
  }

  if (kTfLiteOk != interpreter->Invoke()) {
    TF_LITE_REPORT_ERROR(error_reporter, "Invoke failed.");
  }
  
  TfLiteTensor* output = interpreter->output(0);

  int8_t all_score = output->data.int8[kAllIndex];
  int8_t no_all_score = output->data.int8[kNotAllIndex];
  toggle(all_score, no_all_score, images[6]);
  delay(2000);
```

Where getImage is using decodeSdFile, the value is then passed to DecodeAndProcessImage from the person detector but without   JpegDec.decodeArray(jpeg_buffer, jpeg_length); as the image is passed to it. Below is the output:

```
8:37:56.486 -> Im099_0.jpg
8:37:56.519 -> ===============
8:37:56.580 -> ALL positive score: -74
8:37:56.698 -> ALL negative score: 74
8:37:56.775 -> 
8:37:56.802 -> Im099_0.jpg
8:38:01.981 -> ===============
8:38:02.018 -> ALL positive score: 127
8:38:02.098 -> ALL negative score: -128
8:38:02.194 -> 
8:38:02.219 -> Im099_0.jpg
8:38:07.457 -> ===============
8:38:07.519 -> ALL positive score: -40
8:38:07.623 -> ALL negative score: 40
8:38:07.684 -> 
8:38:07.714 -> Im099_0.jpg
8:38:12.935 -> ===============
8:38:12.997 -> ALL positive score: -74
8:38:13.075 -> ALL negative score: 74
8:38:13.152 -> 
8:38:13.179 -> Im099_0.jpg
8:38:18.426 -> ===============
8:38:18.482 -> ALL positive score: -74
8:38:18.578 -> ALL negative score: 74
8:38:18.657 -> 
8:38:18.685 -> 
```

This then stays consistent, it is correctly identifying a negative Acute Lymphoblastic image, but why the change at the beginning 127, -128, -40, 40?

Next I pass it another negative image:

```
8:42:55.970 -> Im101_0.jpg
8:42:56.021 -> ===============
8:42:56.086 -> ALL positive score: -74
8:42:56.202 -> ALL negative score: 74
8:42:56.292 -> 
8:42:56.322 -> Im101_0.jpg
8:43:01.479 -> ===============
8:43:01.546 -> ALL positive score: 127
8:43:01.635 -> ALL negative score: -128
8:43:01.724 -> 
8:43:01.754 -> Im101_0.jpg
8:43:06.969 -> ===============
8:43:07.055 -> ALL positive score: 0
8:43:07.136 -> ALL negative score: 0
8:43:07.223 -> 
[... 140 lines cut ...]
9:9:58.847 -> ALL positive score: -21
9:9:58.994 -> ALL negative score: -24
9:9:59.122 -> 
9:9:59.155 -> Im041_0.jpg
9:10:01.353 -> ===============
9:10:01.430 -> ALL positive score: 14
9:10:01.529 -> ALL negative score: 6
9:10:01.609 -> 
9:10:01.640 -> Im099_0.jpg
9:10:03.858 -> ===============
9:10:03.932 -> ALL positive score: -46
9:10:04.053 -> ALL negative score: -22
9:10:04.170 -> 
9:10:04.170 -> Im095_0.jpg
9:10:06.373 -> ===============
9:10:06.445 -> ALL positive score: -33
9:10:06.595 -> ALL negative score: -38
9:10:06.720 -> 
9:10:06.755 -> Im101_0.jpg
9:10:08.928 -> ===============
9:10:08.997 -> ALL positive score: -17
9:10:09.110 -> ALL negative score: -14
```

Hoping someone can shed some light lol, again this model in Python classifies everything correct out of this test set, 20 images all together.

AdamMiltonBarker, 2021-07-16:
@advaitjain hi mate could you shed some light one where I may be going wrong here?

AdamMiltonBarker, 2021-07-16:
Full code unfortunately cannot share the dataset you have to apply for it  https://homes.di.unimi.it/scotti/all/ 

```
#include "Arduino.h"
#include <SPI.h>

#include <TensorFlowLite.h>

#include "main_functions.h"
#include "all_model.h"
#include "model_settings.h"

#include "tensorflow/lite/micro/micro_error_reporter.h"
#include "tensorflow/lite/micro/micro_interpreter.h"
#include "tensorflow/lite/micro/micro_mutable_op_resolver.h"
#include "tensorflow/lite/schema/schema_generated.h"
#include "tensorflow/lite/version.h"

#include <JPEGDecoder.h>

String images[]={
    "Im006_1.jpg",
    "Im020_1.jpg",
    "Im024_1.jpg",
    "Im026_1.jpg",
    "Im028_1.jpg",
    "Im031_1.jpg",
    "Im035_0.jpg",
    "Im041_0.jpg",
    "Im047_0.jpg",
    "Im053_1.jpg",
    "Im057_1.jpg",
    "Im060_1.jpg",
    "Im063_1.jpg",
    "Im069_0.jpg",
    "Im074_0.jpg",
    "Im088_0.jpg",
    "Im095_0.jpg",
    "Im099_0.jpg",
    "Im101_0.jpg",
    "Im106_0.jpg"
};

int tp = 0;
int fp = 0;
int tn = 0;
int fn = 0;

namespace {
  tflite::ErrorReporter* error_reporter = nullptr;
  const tflite::Model* model = nullptr;
  tflite::MicroInterpreter* interpreter = nullptr;
  TfLiteTensor* input = nullptr;
  constexpr int kTensorArenaSize = 136 * 1024;
  static uint8_t tensor_arena[kTensorArenaSize];
} 

void setup() {
  
  Serial.begin(9600);
  while (!Serial) {
    ; 
  }

  Serial.println(F("Initialising SD card..."));
  if (!SD.begin(10)) {
    Serial.println(F("Initialisation failed!"));
    return;
  }
  Serial.println(F("Initialisation done."));
[... 356 lines cut ...]
18:7:51.727 -> 
18:7:51.765 -> Im099_0.jpg
18:7:53.755 -> ===============
18:7:53.831 -> ALL positive score: -46
18:7:53.969 -> ALL negative score: -22
18:7:54.101 -> True Negative
18:7:54.195 -> 
18:7:54.235 -> Im101_0.jpg
18:7:56.305 -> ===============
18:7:56.343 -> ALL positive score: -17
18:7:56.478 -> ALL negative score: -14
18:7:56.614 -> True Negative
18:7:56.685 -> 
18:7:56.720 -> Im106_0.jpg
18:7:58.832 -> ===============
18:7:58.935 -> ALL positive score: -42
18:7:59.072 -> ALL negative score: -45
18:7:59.218 -> False Positive
18:7:59.316 -> 
18:7:59.362 -> True Positives: 7
18:8:00.961 -> False Positives: 6
18:8:01.059 -> True Negatives: 4
18:8:01.164 -> False Negatives: 3
```
The classifications are the same each time so at least it is stable :)

advaitjain, 2021-07-16:
This looks like something with the Arduino specific bits which unfortunately we won't be able to help much with debugging.

Some pointers that may help:

 * The kernels are tested on Cortex-M and should be bit exact with x86 so very often, discrepancies between x86 and a target platform are due to bugs / differences external to the NN inference.

 * It may be useful for your debugging to have a standalone binary with the inputs baked into it and run the exact same on code on both x86 and the Arduino. Something like the [person_detection_test](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/examples/person_detection/person_detection_test.cc)

 * You can also try running this same binary on the STM32F4 platform with Renode: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/ci_build/test_stm32f4.sh

    Something like
    ```
    make -f tensorflow/lite/micro/tools/make/Makefile OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET=stm32f4 third_party_downloads
    make -f tensorflow/lite/micro/tools/make/Makefile OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET=stm32f4 test_person_detection_test
    ```

 * What version of the Arduino TFLM library are you using?

AdamMiltonBarker, 2021-07-16:
Thanks for the reply. I will look into your sugestions, by renode do you mean the following ? https://github.com/renode/renode 

The version of TFLM is 2.4.0 APLHA

advaitjain, 2021-07-16:
TFLM already makes use of renode. The following command:
```
make -f tensorflow/lite/micro/tools/make/Makefile OPTIMIZED_KERNEL_DIR=cmsis_nn TARGET=stm32f4 test_person_detection_test
```
will run the test using renode.

It might be worthwhile to use the latest TFLM Arduino library (built from tip of tree)

@petewarden can likely help with that.

AdamMiltonBarker, 2021-07-16:
Thanks this may seem like a dumb question, but other than making basic MQTT IoT device with Arduino and ESP, I really don't have much experience with Arduino. 

 I am understanding I would modify test_person_detection_test to reflect the code in my project, but what additional steps would be required and where would I execute the make command ? The makefile and optimized kernal dir, are they project specific or need modification to match the project? Sorry noob questions. 

Sure I would be happy to test out the latest TFLM. 

I am in the process of setting up the repository for this project, I figured it was a good start at exploring Arduino for ML, I am planning on publishing it to the Microcontroller and Eyes on Edge challenges though so would be great if I could resolve it.

AdamMiltonBarker, 2021-07-17:
Just to update on this, to make sure I retrained the classifier on 2 separate machines and they all receive the same results regards metrics and classification so it definitely is something related to the Arduino / TFLM. I will do the tests you suggested today. Thanks for the advice.

AdamMiltonBarker, 2021-07-18:
@advaitjain himate any documentation on modifying the make file to include a new project ?

AdamMiltonBarker, 2021-07-18:
@advaitjain any documentation on making a custom project and adding it to the makefile, not as straight forward as just making the example based on person_detection_test is it! 

I have downloaded the person detector zip specified in the makefile / download shell file. If my understanding is correct to get the make sorted I will do the following:

 - Rename the downloaded zip to match my project and rename the files inside the zip.
 - Modify the files in the zip IE add positive and negative example to xxx_image_data.cc and xxx_no_image_data.cc and their header files
 - Replace the model and info in the model_data files and header 
 - Have my installation scripts shift this folder to the downloads folder in the make directory
 - Make a copy of the example folder and update all the files in that folder to match my project
 - Run makefile

Is the only way to do this ?

AdamMiltonBarker, 2021-07-19:
Version one is published, https://github.com/AMLResearchProject/ALL-Arduino-Nano-33-BLE-Sense-Classifier In version two I will include the testing above in the documentation and hopefully can get to the bottom of the misclassification on Arduino.

AdamMiltonBarker, 2021-07-19:
Hi @advaitjain sorry to pester I appreciate you are busy. Are the steps above correct for adding a new project to the tests? It is a little time sensitive :D

advaitjain, 2021-07-19:
You are on the right track. Following the same pattern as the existing examples is the way to go.

The person_detection example setup is a bit more complex because we did not want to commit the large images and model files into the repo. For the purposes of testing, you should be able to directly have all the model and inputs be in the source tree, similar to [hello_word](https://github.com/tensorflow/tflite-micro/tree/tensorflow/lite/micro/examples/hello_world).

Alternately, take a look at the [person_detection_benchmark](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/benchmarks/person_detection_benchmark.cc).

Once you have the example / benchmark building for x86, you should be able to also build and run the same binary on stm32f4 without changing any code. That would be the easiest way to verify if the discrepancy is in the TFLM code or in the Arduino-specific bits.

AdamMiltonBarker, 2021-07-19:
OK thank you for the information. I will get to this ASAP. Thanks again.

petewarden, 2021-07-26:
From reading the conversation, it looks like you've managed to get past the initial issue that started this bug? If not, please update with more information, I'm closing this for now.

Thanks for the heads up about the broken links too. I'm working with O'Reilly to try to update their URL shortener to ensure the book links are fixed at least.

closed 2021-07-26 by petewarden