tensorflow/tflite-micro #294: Fix esp32 ci build

mocleiri, 2021-07-17:
I filed #292 to add a community build for esp32.  

When the ci script runs it fails.  Currently the error output can be seen here:
https://github.com/mocleiri/tflite-micro/actions/runs/1040366564

@vikramdattu you fixed it last time (https://github.com/tensorflow/tensorflow/pull/45334) could you take a look?

mocleiri, 2021-07-20:
#292 has been merged.  

The latest failure log is here: https://github.com/tensorflow/tflite-micro/runs/3109849276?check_suite_focus=true

mocleiri, 2021-07-21:
I took an attempt to fix the test_esp32.sh script.  The person detection one is too complicated for me at the moment so I commented it out.  The script can generate the hello world and micro-speech and then build both.

I can see its possible to have the script called from within a docker container so I will figure out that part as it should save quite a bit of download time to just the espressif/idf:v4.3 container instead of cloning the idf and all of its sub modules.

https://github.com/mocleiri/tflite-micro/commit/4d0dd9d35c0b36231d899d5d40cb880b3fd0349c

vikramdattu, 2021-07-21:
Please find this comment as well: https://github.com/tensorflow/tflite-micro/issues/292#issuecomment-883043276 

It could help to fix person detection example if you want to revisit.

advaitjain, 2021-07-21:
I would recommend not creating a docker container at this time. This workflow will run nightly and the total time is less of a concern.

Note that in the next 6 months, we expect to remove project generation support from the TFLM makefile and the recommended path would need for integrations like esp32 and Arduino to have their own GitHub repositories.

That would be a good point in time to optimize the esp32 workflow with a docker container.

mocleiri, 2021-07-22:
Because the espressif idf image is public it was a one liner to switch to building inside of that container.  I just copied what the xtensa f1 build was doing and it worked.

I tried to change person detection based on @vikramdattu's previous commit but I got other errors.  

The other target's seem to be just building the tensorflow-microlite.a instead of the examples so I also added a build for that.

Since you say that the examples will be externalized I think the person detection can be fixed later when that occurs in what ever repo they end up in.

I verified in my fork that the pull request I just made worked:
https://github.com/mocleiri/tflite-micro/runs/3130621531?check_suite_focus=true

advaitjain, 2021-07-23:
I didn't know that a public docker image already existed. Your solution sounds good to me.

PR #310 (2021-07-23): Fix esp32 ci build
BUG=Fixes #294

I was able to run this with some minor alterations on when the job ran on my fork:
https://github.com/mocleiri/tflite-micro/actions/runs/1054903715

Copy from xtensa ci scripts to run the test script inside of a docker
container.

We use the espressif/idf:release-v4.3 image.

Add a build that builds the full tensorflow-microlite.a library.

The person detection example is more complicated and not fixed here.
commit 868b1e73 (2021-07-23): Fix esp32 ci build (#310) (#310)

Fixes #294

Copy from xtensa ci scripts to run the test script inside of a docker
container.

We use the espressif/idf:release-v4.3 image.

Add a build that builds the full tensorflow-microlite.a library.

The person detection example is more complicated and not fixed here.
closed 2021-07-23 by advaitjain