tensorflow/tflite-micro #295: Getting error in Makefile...

dsantiago, 2021-07-17:
Hi,

was trying to follow the TonyML book, but my make calls do not work, it need some header file that is not in the repo.
Folowwing is the error

```bash
make -f tensorflow/lite/micro/tools/make/Makefile hello_world                                                                            2 ✘  base 
find: ../google/: No such file or directory
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/person_model_int8 already exists, skipping the download.
g++ -std=c++11 -fno-rtti -fno-exceptions -fno-threadsafe-statics -Werror -fno-unwind-tables -ffunction-sections -fdata-sections -fmessage-length=0 -DTF_LITE_STATIC_MEMORY -DTF_LITE_DISABLE_X86_NEON -Wsign-compare -Wdouble-promotion -Wshadow -Wunused-variable -Wmissing-field-initializers -Wunused-function -Wswitch -Wvla -Wall -Wextra -Wstrict-aliasing -Wno-unused-parameter -DOSX -DTF_LITE_USE_CTIME -Os -I. -Itensorflow/lite/micro/tools/make/downloads/gemmlowp -Itensorflow/lite/micro/tools/make/downloads/flatbuffers/include -Itensorflow/lite/micro/tools/make/downloads/ruy -Itensorflow/lite/micro/tools/make/downloads/kissfft -c tensorflow/lite//micro/sparkfun_edge/system_setup.cc -o tensorflow/lite/micro/tools/make/gen/osx_x86_64_default/obj/core/tensorflow/lite//micro/sparkfun_edge/system_setup.o
tensorflow/lite//micro/sparkfun_edge/system_setup.cc:22:10: fatal error: 'am_bsp.h' file not found
#include "am_bsp.h"         // NOLINT
         ^~~~~~~~~~
1 error generated.

```

dsantiago, 2021-07-17:
Just to point out, @petewarden need to update the minified links(in the book) from the O'Reilly's side...

advaitjain, 2021-07-19:
This target is covered by our CI, which is currently green. I would guess some issue in you setup.

Can you try the following:
```
make -f  tensorflow/lite/micro/tools/make/Makefile clean clean_downloads
make -f  tensorflow/lite/micro/tools/make/Makefile third_party_downloads TARGET=sparkfun_edge
make -f  tensorflow/lite/micro/tools/make/Makefile hello_world TARGET=sparkfun_edge
```

A related issue might be https://github.com/tensorflow/tflite-micro/issues/199

dsantiago, 2021-07-20:
It was much better now, but still getting an error on last command. I am on Mac, if that helps 

[image]

advaitjain, 2021-07-20:
The error that you are seeing is exactly what was reported on https://github.com/tensorflow/tflite-micro/issues/263

This is not an error that we can reproduce or explain. Please see https://github.com/tensorflow/tflite-micro/issues/263#issuecomment-877313261 for more context. Mac OS is community supported right now. We do not have any continuous integration or ways to test the build on Mac OS. 

My only suggestion here is to try a Linux build via docker by following the instructions from https://github.com/tensorflow/tflite-micro/blob/main/CONTRIBUTING.md#continuous-integration-system and then seeing if that gives you any clues as to what may be happening for you.

If it helps, here is the snippet in the Makefile where the source specialization happens:
https://github.com/tensorflow/tflite-micro/blob/45c80c969820fc96257f79ac197b6391dbbef741/tensorflow/lite/micro/tools/make/Makefile#L571-L588

dsantiago, 2021-07-20:
Ohh understood... at least wee keep registered here if anyone happens to have the same problem...
Thanks for the help.

closed 2021-07-26 by petewarden