tensorflow/tflite-micro #296: Is there optimized kernels for conv_2d in xtensa hifi4 ?

shiyuanren, 2021-07-19:
Does it just support Fusion F1 and Hifi5 optimizing， How can I do in hifi4 dsp
Thanks

advaitjain, 2021-07-19:
The Fusion F1 optimizations support Hifi4 as well. We're going to be changing things to make that clearer in the coming weeks in the documentation, code and makefile commands.

For now, you can run the following command (copied from [here](https://github.com/tensorflow/tflite-micro/blob/4b1b9bd4924614b4ede39d093e639411bba190eb/tensorflow/lite/micro/tools/ci_build/test_xtensa_fusion_f1.sh)):
```
make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads

make -f tensorflow/lite/micro/tools/make/Makefile \
  TARGET=xtensa \
  TARGET_ARCH=fusion_f1 \
  OPTIMIZED_KERNEL_DIR=xtensa \
  XTENSA_CORE=F1_190305_swupgrade \
  test -j8
```

and replace `XTENSA_CORE=<your core>` to use the optimized kernels for your core. (while leaving `TARGET_ARCH=fusion_f1`).

shiyuanren, 2021-07-26:
@advaitjain 
 Thanks,  my problem has been solved.

closed 2021-07-26 by shiyuanren

advaitjain, 2021-07-26:
Please note that with https://github.com/tensorflow/tflite-micro/pull/305, we use `TARGET_ARCH=hifi4` for Fusion F1, Hifi3, Hifi3Z and Hifi4. This matches up with the naming convention for the underlying Xtensa nnlib: https://github.com/foss-xtensa/nnlib-hifi4