tensorflow/tflite-micro #298: Runtime error: Didn't find op for builtin opcode 'REDUCE_PROD' version '1'

mohajjel, 2021-07-20:
Target: esp32

advaitjain, 2021-07-20:
The reason for this is that REDUCE_PROD is currently not implemented for TFLM. If you would like, you can create a PR to port it over from TfLite to TFLM.

Here are some pointers:
 * REDUCE_PROD would need to be added to https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/kernels/reduce.cc
 * Here is the OP porting guide: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/porting_reference_ops.md
   * only a subset of the steps may be relevant here
 * And here is the implementation in TfLite: https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/kernels/reduce.cc

closed 2021-07-26 by petewarden

KamilJamrosMobica, 2021-11-23:
Hello,
I have the same problems. I need REDUCE_PROD, but i can't understand, how can I porting it. I "extract a function for parsing operator". But when i go to Step 2, I don't see Reduce_prod implementation in "tensorflow/lite/kernels/internal/reference/reference_ops.h". 
For this operator, is the implementation not created in the header file but only in the source file?