tensorflow/tflite-micro #302: Failed to get registration from op code AVERAGE_POOL_2D

stanleycelestin1, 2021-07-20:
Getting this error when compiling model that uses average pool 2D.:

```
Didn't find op for builtin opcode 'AVERAGE_POOL_2D' version '2'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

Failed to get registration from op code AVERAGE_POOL_2D
```

advaitjain, 2021-07-20:
If you are using the MicroMutableOpsResolver, you will need to explicitly call AddAveragePool2D.

https://github.com/tensorflow/tflite-micro/blob/45c80c969820fc96257f79ac197b6391dbbef741/tensorflow/lite/micro/micro_mutable_op_resolver.h#L142-L145

See keyword_benchmark as an example https://github.com/tensorflow/tflite-micro/blob/45c80c969820fc96257f79ac197b6391dbbef741/tensorflow/lite/micro/benchmarks/keyword_benchmark.cc#L54-L58

stanleycelestin1, 2021-07-20:
Thanks @advaitjain! I actually had that in my code but my actual issue was that I allocated 5 operations in ` static tflite::MicroMutableOpResolver<5> micro_op_resolver;` but added 7 operations. The last two added operations didn't get captured. Once I changed to `<7>` I no longer have the issue. Thanks for the swift response.

closed 2021-07-20 by stanleycelestin1