tensorflow/tflite-micro #304: Remove runtime code paths for big endian machines

advaitjain, 2021-07-20:
The binary footprint of TFLM has been creeping up and we are in a phase where we would like to reduce that and push functionality into offline pre-processing or brnaches that can be pruned by the compiler/linker.

One such example is big endian support.
https://github.com/tensorflow/tflite-micro/blob/9d1f67e41aadfa55ac6b8bbf6038402ce22ee1e8/tensorflow/lite/micro/micro_allocator.cc#L369-L390

The recommendation for TFLM users that would like big endian support is to have a pre-processing tool that converts the flatbuffer to have the same endianess as the target platform.

We would be happy to discuss alternate strategies if this suggestion is not going to work in practice. However, we would like to have users not rely on the current way of supporting big-endian.

advaitjain, 2021-07-20:
tagging @kun-lu20 and @skribm9

advaitjain, 2021-08-23:
In addition to impact on the TFLM footprint (which we need some testing to validate if indeed there is any), there is also an impact on readability and maintainability of the code.

For example, if we go down the path of https://github.com/tensorflow/tflite-micro/issues/456 then we risk introducing a fair amount of complexity.

closed 2021-11-17 by advaitjain

mariannems, 2022-03-11:
Following this, is there any guidelines to run the TFLM examples on big endian devices ?

mariannems, 2022-03-17:
I would be interested into working on that, but I still have some concerns.

I see how to make a tool that would byte swap the buffers inside the model. That could be done with something similar to [randomize_weights.py](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/tools/randomize_weights.py). It could open the .tflite, loop through the buffers and byte swap them as needed according to the TensorType, and then write the new .tflite using the flatbuffer_utils functions.

But, if the support for big endian is removed from tflm, does it means that the tool would also need to swap all the other values, such as indices or lengths, and other parameters that are written as int32 ? If so, I don't see how it could be made easily. For example, `model_object.Pack(builder)` would generate indices, which would not have been byte swapped.

Any thoughts on that ? Thanks

joshortner, 2023-09-25:
@mariannems Is there a solution to the issue you raised? I'm working on a big endian machine, used the functions in flatbuffer_utils.py to convert the model, but it seems like some of the indices in the model weren't converted causing asserts to trigger. Maybe I'm missing something simple.

kun-lu20, 2023-09-26:
@joshortner Probably the code changes in PR https://github.com/tensorflow/tensorflow/pull/60190 would be helpful.

joshortner, 2023-09-26:
@mariannems Thanks! It seems like the python scripts only byte swaps part of the model buffer. When loading the model on the MCU I'm seeing huge indices that are a result of an integer not being swapped. If you haven't seen the same issues, maybe it's a bug or misuse on my end.

mariannems, 2023-09-26:
@joshortner If ever this can be of any help, here is the script we are using https://github.com/mariannems/tflite-micro/blob/5659d43c8256b87f81c14f39ee4e872b746dc416/tensorflow/lite/tools/flatbuffers_endianness.py

PR #689 (2021-11-17): Remove support for big-endian in the TFLM inference code.
Moving forward, we will require that the endianness of the flatbuffer is the same as the endianness of the system that TFLM is being used for inference on.

BUG=Fixes #304 and http://b/194224890
commit ecf05497 (2021-11-17): Remove support for big-endian in the TFLM inference code. (#689)

Moving forward, we will require that the endianness of the flatbuffer is
the same as the endianness of the system that TFLM is being used for
inference on.

BUG=Fixes #304 and http://b/194224890