tensorflow/tflite-micro #311: Couldn't open elf file generated by hello world example

iams07, 2021-07-22:
Hey, so I was attempting to build the hello world example for RISC-V target and then use Spike to run the subsequent binary file.
These are the commands I used to do this:

1. git clone https://github.com/tensorflow/tflite-micro.git
2. cd tflite-micro
3. Change the target from riscv32_mcu to mcu_riscv in /tensorflow/lite/micro/tools/make/mcu_riscv_makefile.inc and in /tensorflow/lite/micro/examples/hello_world/riscv32_mcu/Makefile.inc
4. Moved the flags `-fnothreadsafe-statics` and `-fno-use-cxa-atexit` from PLATFORM FLAGS to CXX FLAGS in /tensorflow/lite/micro/tools/make/targets/mcu_riscv_makefile.inc
5. Changed -c11 to -gnu11 under CCFLAGS in /tensorflow/lite/micro/tools/make/Makefile
6. Changed the functions fmin and fmax to min and max respectively in /tensorflow/lite/kernels/internal/min.h and /tensorflow/lite/kernels/internal/max.h.
7. Run the command `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=mcu_riscv TARGET_ARCH=riscv32_mcu hello_world_bin`
8. Run the binary using Spike: `spike --isa=rv32imafdc pk tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/hello_world`

I've checked that the binary generated is indeed ELF 32-bit RISC-V executable, but still on running the final command, it shows
`bbl loader
couldn't open ELF program: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/hello_world!
`

Can anyone let me know what I'm doing wrong in this case??

ioannesKX, 2021-07-26:
Similar results here. I follow this process and when I run the binary using Spike it cycles through these (in debug mode):
```
core   0: 0x80000004 (0x34011173) csrrw   sp, mscratch, sp
: 
core   0: 0x80000008 (0x1a010863) beqz    sp, pc + 432
: 
core   0: 0x8000000c (0x04a13823) sd      a0, 80(sp)
core   0: exception trap_illegal_instruction, epc 0x8000000c
core   0:           tval 0x04a13823
```

I tried the person detection example too (changing the target in the relevant Makefile.inc), and this one doesn't build at all, with this error:
```
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/8.1.0/../../../../riscv64-unknown-elf/bin/ld: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/person_detection_int8 section `.bss' will not fit in region `ram'
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/8.1.0/../../../../riscv64-unknown-elf/bin/ld: section .stack VMA [0000000080003800,0000000080003fff] overlaps section .bss VMA [00000000800000d8,000000008002230f]
/opt/riscv/bin/../lib/gcc/riscv64-unknown-elf/8.1.0/../../../../riscv64-unknown-elf/bin/ld: region `ram' overflowed by 0 bytes
collect2: error: ld returned 1 exit status
make: *** [tensorflow/lite/micro/examples/person_detection/Makefile.inc:81: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/person_detection_int8] Error 1
```
I think it has to do with the custom linker scripts from SiFive bsp that are included in the makefiles, maybe the binary is meant for some specific boards and not for Spike.

petewarden, 2021-07-26:
I'm hoping at PR #321 should get RISC V support back to the level it was at a few months ago, which was tested on a SiFive FE310 board. I haven't run this on hardware or an emulator yet though, I've filed #322 to track automating that.

You are right that different boards will probably require different setups, especially on the linker side. Without understanding all the details of the particular hardware, it's tough for us to help debug these kind of issues, so closing this for now.

closed 2021-07-26 by petewarden

iams07, 2021-07-27:
@ioannesKX 
Your first issue may be because you're not using the correct versions of spike and pk, you should configure them in such a way that they are able to run 32-bit binaries, and use the --isa flag for Spike while executing.
For the 2nd issue, you will have to modify the linker file `tensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/env/freedom-e300-hifive1/flash.lds` and increase the ram size manually in there.
I think that this may be meant for a board too, but there are people who have run tflite codes on Spike, but haven't mentioned how to do so, hence my problem

ioannesKX, 2021-07-27:
Thanks for the reply. Is building tflite micro for 64 bit as trivial as changing the flags in the makefile, or do I have to make other changes?

ioannesKX, 2021-07-28:
OK removing the SiFive BSP files from compilation was easy, I managed to run `hello_world` and `person_detection_int8` on Spike RISC-V sim. Compiled for `rv64gc` just by changing the makefile, but I suppose it would work the same for the default `rv32imac`. 

Tell me if you're interested for a pull request for this or to include Spike runtime functionality tests in the CI script @petewarden .

iams07, 2021-07-28:
> OK removing the SiFive BSP files from compilation was easy, I managed to run `hello_world` and `person_detection_int8` on Spike RISC-V sim. Compiled for `rv64gc` just by changing the makefile, but I [...]

Can you share how you ran the hello world example on Spike, what changes did you make in the files etc?? Thanks in advance

ioannesKX, 2021-07-28:
[diff.txt](https://github.com/tensorflow/tflite-micro/files/6893374/diff.txt)
I uploaded the diff for `tensorflow/lite/micro/tools/make/targets/mcu_riscv_makefile.inc`. I just deleted any line related to the sifive folder. There's some similar stuff on each example's makefile but they probably don't play a role.

iams07, 2021-07-28:
Did you only make changes in the mcu_riscv_makefile.inc? I commented out the lines referring to sifive folder and got an error
`tensorflow/lite/micro/tools/make/downloads/sifive_fe310_lib/bsp/libwrap/sys/read.c:8:10: fatal error: platform.h: No such file or directory
`

ioannesKX, 2021-07-28:
Yes I only changed that file. I had also changed `-mabi` and `-march` to target `rv64gc` (with riscv toolchain I built) but I tried with the default `rv32imac` (with the prebuilt toolchain the makefile downloads) just now and it works the same. You might have leftovers from previous builds, maybe try make clean.

iams07, 2021-07-28:
I tried it with a fresh clone of tflite-micro and it is still not working for me, even without removing the Sifive lines. Can you give the exact steps you followed to get the build running, from the start. I'm very sorry if I'm bothering you, it's just I'm stuck and everytime some new problem comes up

ioannesKX, 2021-07-28:
1) clone repository
2) use this as `micro/tools/make/targets/mcu_riscv_makefile.inc`: https://pastebin.com/BgqEq5i7 (noticed i removed the downloads too, you should have a riscv compiler in your path)
3) `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=mcu_riscv TARGET_ARCH=riscv32_mcu hello_world_bin`

if it doesn't work it might be a compiler problem, maybe install on path the one that tflite micro downloads

iams07, 2021-07-29:
Thank you so much mate, it works now using `riscv32-unknown-elf-gcc` that I had installed previously, but I noticed that the binaries went into infinite loop of repeating the same output, is that the case with you as well?? I ran it on Spike using the command
`spike --isa=rv32gc pk tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/hello_world`

ioannesKX, 2021-07-29:
Use `--isa=rv32imac` flag on Spike as that's the `-march` used in the makefile.

iams07, 2021-07-29:
Oh ok, thanks again. Yeah, I meant that the actual outputs were repeating, with `--isa=rv32imac` I get an error that `an illegal instruction was executed`, but rv32gc actually seems to be working correctly if there's a while(true) loop, so I'll stick with that then

pruth111, 2021-11-09:
> Hey, so I was attempting to build the hello world example for RISC-V target and then use Spike to run the subsequent binary file. These are the commands I used to do this:
> 
>     1. git clone https://github.com/tensorflow/tflite-micro.git
> 
>     2. cd tflite-micro
> 
>     3. make -f tensorflow/lite/micro/tools/make/Makefile third_party_downloads
> 
>     4. Change the target from riscv32_mcu to mcu_riscv in /tensorflow/lite/micro/tools/make/mcu_riscv_makefile.inc and in /tensorflow/lite/micro/examples/hello_world/riscv32_mcu/Makefile.inc
> 
>     5. Moved the flags `-fnothreadsafe-statics` and `-fno-use-cxa-atexit` from PLATFORM FLAGS to CXX FLAGS in /tensorflow/lite/micro/tools/make/targets/mcu_riscv_makefile.inc
> 
>     6. Changed -c11 to -gnu11 under CCFLAGS in /tensorflow/lite/micro/tools/make/Makefile
> 
>     7. Changed the functions fmin and fmax to min and max respectively in /tensorflow/lite/kernels/internal/min.h and /tensorflow/lite/kernels/internal/max.h.
> 
>     8. Run the command `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=mcu_riscv TARGET_ARCH=riscv32_mcu hello_world_bin`
> 
>     9. Run the binary using Spike: `spike --isa=rv32imafdc pk tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/hello_world`
> 
> 
> I've checked that the binary generated is indeed ELF 32-bit RISC-V executable, but still on running the final command, it shows `bbl loader couldn't open ELF program: tensorflow/lite/micro/tools/make/gen/mcu_riscv_riscv32_mcu_default/bin/hello_world! `
> 
> Can anyone let me know what I'm doing wrong in this case??

I am getting the same issue, Apart from compiling the executable locally(using the toolchain built locally), is there a possible way to get it running via Sifive toolchain? I just wanted to know the reason for not being able to recognize by spike. If anyone could let me know , it would be very helpful. Thanks !

pruth111, 2021-11-10:
> Hey, so I was attempting to build the hello world example for RISC-V target and then use Spike to run the subsequent binary file. These are the commands I used to do this:
>
>     1. git clone [...]

Hello Team, I am facing the exact issue after building the hello world for the riscv. May i know the reason for not able to run on Spike? Is building the toolchain locally and then building the tflite micro the only solution?

ahwuyeah, 2022-10-09:
I modify `tensorflow/lite/micro/tools/make/Makefile` and placed it  at the folder root of tflite-micro.
And it could support  RISC-V with rv64gv and no need to add flag when simulating with spike.
This is my [Makefile](https://github.com/ahwuyeah/spike_tflite-micro/blob/main/Makefile).
At beginning, please follow [Readme](https://github.com/ahwuyeah/spike_tflite-micro/blob/main/README.md) to recompile your RISC-V toolchain.

Bruno-Lovison-Franco, 2024-01-08:
> 1. clone repository
> 2. use this as `micro/tools/make/targets/mcu_riscv_makefile.inc`: https://pastebin.com/BgqEq5i7 (noticed i removed the downloads too, you should have a riscv compiler in your [...]

Thanks, your makefile flags helped me to compile for my CV32E40P.