tensorflow/tflite-micro #312: No make tool when installing IDF using installer on Windows.

jam244, 2021-07-22:
Hi , 
I'm new to Tflite micro. I am attempting to build the hello_world example for ESP32 on Windows. 
The Readme https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/hello_world says "The sample has been tested on ESP-IDF version 4.0 " 
But the IDF docs https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started-legacy/windows-setup.html says "Since ESP-IDF V4.0, the default build system is based on CMake." 
The Readme then proceeds to use "make" to generate the project. Make is not available when installing IDF using installer.
So, I tried installing the  "Legacy GNU Make" setup for the toolchain and IDF but then it doesn't have the option to run the example using idf.py.

Can the steps taken to deploy the hello_world to the ESP32 target be elaborated more so that it can replicated easily.

mocleiri, 2021-07-23:
This question was also asked in the sig-micro glitter.  The problem was missing the generate step which produces the IDF 4.3 compatible project.

Also there were issues building using the windows IDF.

The example did generate and build right when using Linux.

advaitjain, 2021-07-23:
I'm going to close this issue since it sounds like this expected from the standpoint of TFLM. Espressif forums would likely be a better place for such a feature request.

closed 2021-07-23 by advaitjain

jam244, 2021-07-23:
Adding a line in the readme specifying that validation is only done on Linux would save time for developers.

advaitjain, 2021-07-23:
That's a good idea. Would you be up for sending a PR with that change?