tensorflow/tflite-micro #329: Incorrect convolution computation in a specific case on Xtensa HiFi4

lvadke, 2021-07-27:
Hi, I'm not sure if TFLM supports Xtensa HiFi4 cores, because the optimized kernels have preprocessing conditions like:
```c++
#if defined(HIFIMINI)
      ConvEvalHifiMini(ConvParamsQuantized(params, op_data.reference_op_data),
                       op_data.reference_op_data.per_channel_output_multiplier,
                       op_data.reference_op_data.per_channel_output_shift,
                       tflite::micro::GetTensorShape(input),
                       tflite::micro::GetTensorData<int8_t>(input),
                       tflite::micro::GetTensorShape(filter),
                       tflite::micro::GetTensorData<int8_t>(filter),
                       tflite::micro::GetTensorShape(bias),
                       tflite::micro::GetTensorData<int32_t>(bias),
                       tflite::micro::GetTensorShape(output),
                       tflite::micro::GetTensorData<int8_t>(output));
#elif defined(FUSION_F1) || defined(HIFI5)
      ConvEvalHifi(context, node, params, op_data, input, filter, bias, output);
#else
      return ConvReferenceEvalInt8(context, node);
#endif
```
But if it does (I used a small workaround), then it computes convolution incorrectly in one specific case. When I compare the outputs from HiFi4 and reference implementations, I see some differences:

[image]

I created a unit test which reproduces the issue.

[conv_test.txt](https://github.com/tensorflow/tflite-micro/files/6887125/conv_test.txt)

On Xtensa Fusion F1 core the computation works fine.

advaitjain, 2021-07-27:
The hifimini implementation is not intended for use with Hifi4.

The naming was unclear previously but that has recently been improved. Please see this comment for the current way to build TFLM with Hifi4 optimized kernels:
https://github.com/tensorflow/tflite-micro/issues/296#issuecomment-886868989

lvadke, 2021-07-27:
Sorry, I was not clear which configuration I used, but I used the same for HiFi4 as for Fusion F1. HiFi Mini would not compile with the HiFi4 compiler configuration (different inline assembler). I stepped through the code and it uses the HiFi4 NN library.

nyadla-sys, 2021-07-27:
@cad-audio

advaitjain, 2021-07-27:
Thanks for the clarification. We'll look into this along with the nnlib authors (@cad-audio)

cad-audio, 2021-07-27:
@advaitjain, below PR should resolve this issue.
https://github.com/tensorflow/tflite-micro/pull/330

advaitjain, 2021-07-27:
@lvadke, if you can confirm that your issue is resolved once #330 is merged, that would be great.

Please note that to get the updated xannlib, you will need to delete what is currently downloaded prior to building the code:
```
rm -rf tensorflow/lite/micro/tools/make/downloads/xa_nnlib_hifi4
```

I'm going to leave this github issue open until we add a new test case to the conv kernel tests based on the file attached with https://github.com/tensorflow/tflite-micro/issues/329#issue-954051795

lvadke, 2021-07-28:
Yes, update of the HiFi4 NN library to version 2.4.1 fixed the issue. Thank you for very quick resolution.

nyadla-sys, 2021-08-03:
@lvadke Closing this issue as it is working as expected with 2.4.1 lib

closed 2021-08-03 by nyadla-sys