tensorflow/tflite-micro #331: How can I create a MicroMutableOpResolver with a size determined at runtime?

mocleiri, 2021-07-27:
For my micropython implementation of the tflite-micro c++ api I wanted to create a way  to use the MicroMutableOpResolver so that users could reduce the memory overhead versus using the AllOpsResolver.

I'm having trouble getting it to compile because the number of ops the resolver will take is specified as a template parameter instead of an argument to the constructor of the class.

I have a custom micropython object to represent the op resolver and it takes the number of ops the user expects to register but it gets this value at runtime and then I try to use to create the MicroMutableOpResolver which fails.

```
../../../micropython-modules/microlite/openmv-libtf.cpp: In function ‘int libtf_init_op_resolver(microlite_op_resolver_obj_t*)’:
../../../micropython-modules/microlite/openmv-libtf.cpp:65:53: error: the value of ‘number_of_ops’ is not usable in a constant expression
         tflite::MicroMutableOpResolver<number_of_ops>* op_resolver = new tflite::MicroMutableOpResolver<number_of_ops>(&micro_error_reporter);
                                                     ^
../../../micropython-modules/microlite/openmv-libtf.cpp:63:28: note: ‘number_of_ops’ was not initialized with a constant expression
         const unsigned int number_of_ops = microlite_op_resolver->number_of_ops;
                            ^~~~~~~~~~~~~
```

https://github.com/mocleiri/tensorflow-micropython-examples/blob/6bc25a90db22c109200e82667cd11ef658b21336/micropython-modules/microlite/openmv-libtf.cpp#L65

Is there a way I can create an instance of the tflite::MicroMutableOpResolver using a dynamic size or should I look at subclassing and making my own op resolver?

petewarden, 2021-07-28:
Hi Michael,
                   unfortunately the op resolver reduces memory overhead by helping the linker remove unused code. Because an op resolver that worked at runtime couldn't rely on this dead-code stripping, it would need to have all of the implementations available, and so would be identical to AllOpsResolver in memory footprint. Does that make sense? I'm happy to have a quick call to discuss this if it would be helpful, since I know this is an important topic.

mocleiri, 2021-07-29:
I wasn't aware of the [dead code stripping feature.](https://stackoverflow.com/a/6770305)  The template versus runtime field makes sense in that context.

For my use-case there are two memory sizes:
1. Firmware size
2. System Runtime memory use

In the future I may be able to address 1 using micropython native modules or setup compile time options to let the end-user control which ops are built into the firmware at build time.

At the moment I'm starting at 2.  Can I reduce system memory use at runtime by using the MicroMutableOpResolver instead of the AllOpsResolver.

It looked like I could save 128(- number of ops used)*sizeof(TfLiteRegistration) plus any memory associated to the options objects that each op instantiates when registered.

I will make a subclass and evaluate if there is any runtime memory improvement.

closed 2021-07-29 by mocleiri