tensorflow/tflite-micro #332: Zephyr sample build broken

laurenmurphyx64, 2021-07-27:
**System information**

- Have I written custom code (as opposed to using a stock example script provided in TensorFlow): No
- OS Platform and Distribution: Linux Ubuntu 18.04
- TensorFlow installed from (source or binary): source
- GCC/Compiler version: Ubuntu 7.5.0-3ubuntu1~18.04

**Describe the current behavior**

After the Zephyr build system compiles all of the TFLM library files itself based on the `CMakeLists.txt` generated by TF's build system, it breaks during linking with undefined references:

```
...
[ 94%] Linking CXX executable zephyr_prebuilt.elf
...
~/zephyr-sdk-0.12.3/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: ../app/libapp.a(all_ops_resolver.cc.obj): in function `tflite::MicroMutableOpResolver<128u>::AddArgMin()':
~/tflite-micro/tensorflow/lite/micro/tools/make/gen/zephyr_vexriscv_x86_64_default/prj/hello_world/cmake/./tensorflow/lite/micro/micro_mutable_op_resolver.h:138: undefined reference to `tflite::Register_AVERAGE_POOL_2D()'
...
(74 other undefined references)
```

`libtensorflow-microlite.a`, which TF's build system builds but the Zephyr build system does not use, compiles and links successfully. 

Comparing the files included in the `libtensorflow-microlite.a` (from the log) to the files listed in the generated `tensorflow/lite/micro/tools/make/gen/zephyr_vexriscv_x86_64_default/prj/hello_world/cmake/third_party/CMakeLists.txt`, my conclusion is that Zephyr is missing the files of `MICROLITE_CC_KERNEL_SRCS` (listed at `tensorflow/lite/micro/tools/make/Makefile:349`):

```
MICROLITE_CC_KERNEL_SRCS := \
tensorflow/lite/micro/kernels/activations.cc \
tensorflow/lite/micro/kernels/activations_common.cc \
tensorflow/lite/micro/kernels/add.cc \
...
```

**Describe the expected behavior**

The Zephyr build of the samples should be able to link the executable without undefined references.

**[Contributing](https://www.tensorflow.org/community/contribute)**

- Do you want to contribute a PR? (yes/no): Yes
- Briefly describe your candidate solution(if contributing):

Change `tensorflow/lite/micro/examples/hello_world/zephyr_riscv/Makefile.inc:15` and `tensorflow/lite/micro/examples/magic_wand/zephyr_riscv/Makefile.inc:18` from

```
$(eval $(call generate_project,cmake,zephyr_cmake_project.cmake,hello_world,$(MICROLITE_CC_SRCS) $(THIRD_PARTY_CC_SRCS) $(ZEPHYR_HELLO_WORLD_SRCS) $(MICROLITE_CC_HDRS) $(THIRD_PARTY_CC_HDRS) $(HELLO_WORLD_HDRS),,$(LDFLAGS) $(MICROLITE_LIBS),$(CXXFLAGS),$(CCFLAGS),))
```

to:

```
$(eval $(call generate_project,cmake,zephyr_cmake_project.cmake,hello_world,$(MICROLITE_CC_SRCS) $(MICROLITE_CC_KERNEL_SRCS) $(THIRD_PARTY_CC_SRCS) $(ZEPHYR_HELLO_WORLD_SRCS) $(MICROLITE_CC_HDRS) $(THIRD_PARTY_CC_HDRS) $(HELLO_WORLD_HDRS),,$(LDFLAGS) $(MICROLITE_LIBS),$(CXXFLAGS),$(CCFLAGS),))
```

(Add `$(MICROLITE_CC_KERNEL_SRCS)`)

**Standalone code to reproduce the issue**

`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=zephyr_vexriscv hello_world_bin`

**Other info / logs** 

See comment.

mocleiri, 2021-07-28:
@laurenmurphyx64 you should consider adding a community supported zephyr build.  It can build the library and the examples daily.

I made one for esp32 last week in #293.  

You put a script into tensorflow/lite/micro/tools/ci_build that runs the build and then add a GitHub action to invoke it. It's status is shown on the readme and helps detect breakage sooner.

petewarden, 2021-07-28:
Just to back up @mocleiri's comment, we'd love to have a community build tracking the Zephyr integration. #293 is a good template to follow, and I've got some similar PRs for RISC-V linked from #322.

advaitjain, 2021-07-28:
Note that TFLM is moving towards encouraging targets to build examples in their own standalone reporitories.

@laurenmurphyx64 , can you try out https://github.com/antmicro/tensorflow-zephyr-vexriscv-examples?

My preference would be to delete zephyr integration from tflite-micro and have everyone use https://github.com/antmicro/tensorflow-zephyr-vexriscv-examples

laurenmurphyx64, 2021-07-28:
> Note that TFLM is moving towards encouraging targets to build examples in their own standalone repositories.
> 
> @laurenmurphyx64 , can you try out https://github.com/antmicro/tensorflow-zephyr-vexriscv-examples?
> 
> My preference would be to delete zephyr integration from tflite-micro and have everyone use https://github.com/antmicro/tensorflow-zephyr-vexriscv-examples

@advaitjain That's cool with me! 👍 I'd prefer it too. But if you delete the zephyr integration from tflite-micro, I think the Antmicro repository will no longer work. Their [build script](https://github.com/antmicro/tensorflow-zephyr-vexriscv-examples/blob/main/build.sh) uses it to build:

```
...
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=zephyr_vexriscv $1_bin
...
```

The [Zephyr repo](https://github.com/zephyrproject-rtos/zephyr/) recently added [two samples](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/modules/tensorflow) and [official build support](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/tensorflow/CMakeLists.txt) for TensorFlow Lite Micro as a library and doesn't use the TFLM build system. So, the Zephyr repo will be fine even after integration is removed from the TFLM repo. More details if you're curious: https://github.com/zephyrproject-rtos/zephyr/pull/34057.

I've used the Antmicro repo before and I like it a lot. I referenced their [old repository](https://github.com/antmicro/litex-vexriscv-tensorflow-lite-demo) while adding the build support and samples (one has Renode support) to Zephyr. But I'd prefer if Zephyr users were pointed towards the Zephyr repo first now that the support is there.

advaitjain, 2021-07-28:
Happy to get your current fix merged (we get get that in once the CLA is signed).

Adding in @PiotrZierhoffer from Antmicro to this conversation. TFLM won't delete zephyr integration from tflite-micro without a transition plan for the Antmicro repository.

However, the generate_project macro invocation that you are fixing with https://github.com/tensorflow/tflite-micro/pull/333/ is hard to maintain and untested and may again be broken by changes to TFLM.

My primary concern is to make sure that updates to TFLM are less likely to break the Zephyr build. To that end, we are rolling out a [new project generation interface](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py).

If you decide to use this project generation interface, then the process of updating tflite-micro in Zephyr would look like:
```
git clone https://github.com/tensorflow/tflite-micro.git
cd tflite-micro
python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py <path_to_tflite-micro_in_zephyr>
```

The file paths in CMakeLists would still have to be manually updated. Or you could try and use something like:
```
python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py <path_to_tflite-micro_in_zephyr> --print_dest_files --no_copy
```

This create_tflm_tree.py is tested in our CI and the expectation is that it will make it much easier to stay up to date with the TFLM code.

Forking the TFLM samples and having the documentation be Zephyr specific is the way to go.

PiotrZierhoffer, 2021-07-28:
Cc @mateusz-holenko

I'll be able to get back to this next week

laurenmurphyx64, 2021-08-02:
@advaitjain Got myself covered by the CLA. Thanks for letting me know about the new project generation template! I'll look into  using it in Zephyr.

closed 2021-08-02 by advaitjain

videvelop, 2023-10-09:
> Note that TFLM is moving towards encouraging targets to build examples in their own standalone reporitories.
>
> @laurenmurphyx64 , can you try out [...]

==
This sample is broken for the current tflite_micro repo. Are there any other updated samples available?
https://github.com/tensorflow/tflite-micro

videvelop, 2023-10-09:
> Forking the TFLM samples and having the documentation be Zephyr specific is the way to go.

==
Anything done on the following?
_Forking the TFLM samples and having the documentation be Zephyr specific is the way to go._

If not, is there any change in the thought process?

PR #333 (2021-08-02): Fix example Zephyr build
BUG=Fixes #332 

On commit `5c7c11e` of the tensorflow repo at `tensorflow/lite/micro/tools/make/Makefile:664`, `MICROLITE_CC_SRCS` contains `MICROLITE_CC_KERNEL_SRCS`. In the new tflite-micro repo, it does not contain `MICROLITE_CC_KERNEL_SRCS`. This commit adds `MICROLITE_CC_KERNEL_SRCS` to
`tensorflow/lite/micro/examples/hello_world/zephyr_riscv/Makefile.inc` so that the generated Zephyr `CMakeLists.txt` doesn't miss these files.

Similar fix made to corresponding file for `magic_wand` sample.