tensorflow/tflite-micro #335: Oreilly TinyML example build issue.

nemo8, 2021-07-28:
Oreilly TinyML example build issue: page 84 
OS Windows x64 (cygwin)

make -f tensorflow/lite/micro/tools/make/Makefile test_hello_world_test

Log:
$ make -f tensorflow/lite/micro/tools/make/Makefile test_hello_world_test
FIND: Parameter format not correct
FIND: Parameter format not correct
File not found - *.cc
File not found - *.h
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/person_model_int8 already exists, skipping the download.
make: *** No rule to make target 'test_hello_world_test'.  Stop.

petewarden, 2021-07-28:
This looks like another instance of Windows build issues, very similar to #170. As @advaitjain mentions there:

> Windows is currently fully community supported and the builds are not part of any CI or something that the TFLM engineers are set up to test. Often errors are the result of some incompatibility with the shell commands (which appears to be the case here).

Is there a particular device you're targeting? IDEs like Arduino and ESP32 do work well for using TFL Micro on Windows, and avoid these issues with make.

nemo8, 2021-07-28:
@petewarden thanks for the comment. I have exactly the same issue as here (Page 84
1st paragraph):  https://www.oreilly.com/catalog/errataunconfirmed.csp?isbn=0636920254508
for this task, I don't have to specify a target (according to the article in the book). 

I will use the device: Arduino Nano 33 BLE Sense: https://store.arduino.cc/arduino-nano-33-ble-sense

petewarden, 2021-07-30:
Have you tried using the Arduino IDE to load the TensorFlow Lite Micro library from the Library Manager? That is tested cross-platform, unlike our make build which isn't guaranteed to run on Windows.

nemo8, 2021-07-30:
I haven't tried it yet.
 I've checked make **-f tensorflow/lite/micro/tools/make/Makefile test_hello_world_test** command on linux, it works fine. So I keep doing all task in linux enviroment.

embeddetech, 2021-08-04:
Hi @nemo8,
I do have a solution for the Windows build issues, I'll be working on a pull request.

nemo8, 2021-08-04:
> Hi @nemo8,
> I do have a solution for the Windows build issues, I'll be working on a pull request.

Hi @embeddetech , thanks for comment. Please let me know when pull request is ready.

AliAqdas-repo, 2021-08-10:
I had some similar issues. I am not sure if yours is the same as mine but I had to install some dependencies to resolve mine. I called the following commands before running the makefile command.
```
sudo apt update
sudo apt-get install zip unzip
sudo apt install gcc
sudo apt install g++
```
There might be some issues installing these so you'd need to install them in the order they are mentioned

nemo8, 2021-08-10:
@AliAqdas-repo  if I use Linux, everything works fine.

embeddetech, 2021-08-18:
Hi @nemo8 , 
Just FYI, I have everything building correctly, and *almost* have the micro speech demo running in Visual Studio and getting fed real system audio and printing to a virtual console terminal in Virtuoso. There's a video I shared in a comment I just posted to #170 in case you wanted to check it out.

Really hoping to get TFLM up and running in Virtuoso so I can go to town with really cool portable examples of TFLM that either require no hardware, or cheap and no-code hardware (like an accelerometer for the magic wand example).

petewarden, 2021-10-04:
Since there are several issues in here, and I believe the original problem is a duplicate of the #170 Windows problems, closing this one. Please do open other issues if you believe this was closed in error.

closed 2021-10-04 by petewarden

sell4fit, 2023-02-23:
Hello,

I am going thru Oreilly TinyML book. trying to run a command don page 84 in Ubuntu Linux:
make -f tensorflow/lite/micro/tool/make/Makefile test_hello_world_test

the path only goes up to "tensorflow/lite/micro/". has the path change for the past year?

thanks

La-guajolota, 2025-01-04:
Hi @sell4fit! I just ran into the same problem. The clone repo isn't the same as supposed in the book anymore... Did you happen to solve this pity?