tensorflow/tflite-micro #339: Output seemingly random values

duncmac16, 2021-07-29:
Hi, I've been trying to use my own model in TFLM for an ESP32, but I have been running into an issue with the output of the model. I can load in the model fine and allocate the tensors without any errors, but I can't seem to get sensible outputs from the output tensor. 

This is the setup I use for the model:
```
void setup_tensorflow()
{
    tflite::InitializeTarget();
    std::cout << "[2.1] setting up error reported" << std::endl;
    error_reporter = &micro_error_reporter;

    model = tflite::GetModel(tflite_model_tflite);
    if (model->version() != TFLITE_SCHEMA_VERSION)
    {
        error_reporter->Report(
            "Model provided is schema version %d not equal "
            "to supported version %d.",
            model->version(), TFLITE_SCHEMA_VERSION);
    }

    std::cout << "[2.2] setting up operations resolver" << std::endl; 
    static tflite::MicroMutableOpResolver<8> resolver;
    resolver.AddConv2D();
    resolver.AddSoftmax();
    resolver.AddMaxPool2D();
    resolver.AddReshape();
    resolver.AddQuantize();
    resolver.AddMul();
    resolver.AddAdd();
    resolver.AddFullyConnected();

    std::cout << "[2.3] setting up interpreter" << std::endl; 
    static tflite::MicroInterpreter static_interpreter(
        model, resolver, tensor_arena, kTensorArenaSize, error_reporter);
    interpreter = &static_interpreter;

    std::cout << "[2.4] Alllocating tensors" << std::endl; 
    TfLiteStatus allocate_status = interpreter->AllocateTensors();
    if (allocate_status != kTfLiteOk)
    {
        error_reporter->Report("AllocateTensors() failed.");
    }

    
    input = interpreter->input(0);
    output = interpreter->output(0);

    std::cout << "[2.5] Input shape: (" << input->dims->data[0] << ", " << input->dims->data[1] 
        << ", " << input->dims->data[2] << ", " << input->dims->data[3] << ")" << std::endl;

    std::cout << "[2.6] Input size in bytes: " << input->bytes << std::endl;
    std::cout << "[2.7] Output size in bytes: " << output->bytes << std::endl;
}
```

This is the code I use to invoke the model, as of now, `mfcc_output` is a 1D array of floats generated by printing some of the training data to a c++ file:
```
void perform_inference(default_mfcc*& mfcc)
{
    if (mfcc) {
        input->data.data = mfcc_output;
    } 
    else {
        for (std::size_t i{0}; i < mfcc->size(); ++i) {
            input->data.f[i] = 0.0f;
        }
    }

    input->data.data = mfcc_output;
    TfLiteStatus invoke_status = interpreter->Invoke();
    if (invoke_status != kTfLiteOk) {
        error_reporter->Report("Invoke failed.");
        return;
    }
}
```
And this is the function to read the output tensor:
```
std::string read_predictions()
{
    if (!output) {
        std::cout << "output not valid" << std::endl;
    }

    const int num_elements = output->dims->data[1];
    int8_t m_quantized = output->data.int8[0];
    float m = (float(m_quantized) - output->params.zero_point) * output->params.scale;
    std::cout << "[" << m * 100.f << ", ";
    std::size_t m_i{0};
    for (std::size_t i{1}; i < num_elements; ++i) {
        int8_t quantized = output->data.int8[i];
        float output_value = (float(quantized) - output->params.zero_point) * output->params.scale;
        std::cout << output_value * 100.f << ", ";
        if (output_value > m) {
            m = output_value;
            m_i = i;
        }
    }

    std::cout << "]" << std::endl;

    return words[m_i];
}
```
This is the model summary for the model being used on the board:
```
Layer (type)                 Output Shape              Param #   
=================================================================
conv2d_2 (Conv2D)            (None, 11, 85, 16)        160       
_________________________________________________________________
dropout_3 (Dropout)          (None, 11, 85, 16)        0         
_________________________________________________________________
max_pooling2d_2 (MaxPooling2 (None, 5, 42, 16)         0         
_________________________________________________________________
batch_normalization_2 (Batch (None, 5, 42, 16)         64        
_________________________________________________________________
conv2d_3 (Conv2D)            (None, 3, 40, 32)         4640      
_________________________________________________________________
dropout_4 (Dropout)          (None, 3, 40, 32)         0         
_________________________________________________________________
max_pooling2d_3 (MaxPooling2 (None, 1, 20, 32)         0         
_________________________________________________________________
batch_normalization_3 (Batch (None, 1, 20, 32)         128       
_________________________________________________________________
flatten_2 (Flatten)          (None, 640)               0         
_________________________________________________________________
dense_4 (Dense)              (None, 64)                41024     
_________________________________________________________________
dropout_5 (Dropout)          (None, 64)                0         
_________________________________________________________________
dense_5 (Dense)              (None, 11)                715       
=================================================================
Total params: 46,731
Trainable params: 46,635
Non-trainable params: 96
_________________________________________________________________
```
And this is the code used to quantize the model and convert it to a tensorflow lite model:
```
converter = tf.lite.TFLiteConverter.from_saved_model(model_dir)
converter.optimizers = [tf.lite.Optimize.DEFAULT]
converter.representative_dataset = representative_dataset
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8]
converter.inference_input_type = tf.float32
converter.inference_output_type = tf.int8
tflite_model = converter.convert()
open(model_dir + "/tflite_model.tflite", "wb").write(tflite_model)
```

Loading the tflite model with python and running inferences with it works as expected. All of the confusion matrices seem to make sense, but when it is loaded onto the ESP the model will start outputting random values even though the input `mfcc_output` stays constant. Is there something obvious that I'm missing?

petewarden, 2021-07-30:
Sorry you're hitting issues! I don't see anything obvious from inspecting the code you've shared. Here are a few things that might help debug what's going wrong:

- Can you try writing to the output->data.int8 before Invoke() is called, to fill it in with a marker value (like -23, say). If the output is still that same value after Invoke, then at least you know the model is not executing. If it is still full of random values, then we'll know it's running, but calculating incorrectly.
- Are you able to share a link to your TFLite file? Ideally it would be helpful to also have a minimal x86 program that loads it, passes in some input values, and logs the results.

duncmac16, 2021-08-03:
I just tried filling the output tensor with known values and it still doesn't seem to be working. I made a [repo](https://github.com/duncmac16/tflite-minimal) with a minimal application.

njeffrie, 2021-08-03:
I took a look and didn't spot anything obvious straight away, but I would suggest some more steps to debug:

- You say the model "still doesn't work", but what do you see when you fill the output tensor with known values? Are the values still there or are they random (i.e. model ran but produced garbage)?
- Can you try converting a full-float version of the model and run that instead of the quantized model? That way we can isolate if this is an issue due to quantization, or something else.
- If the float version does not work either, can you replace your model with a known-good model (for example, hello world) and see if it's an issue with the model or an issue with your application?

duncmac16, 2021-08-03:
- Yes, by "not working" I do mean random outputs on a constant input. After setting all of the values in the output to -23 and running the Invoke method, the output tensor will show mostly random values in random positions in the array. That being said, most of the output tensor array is zero except for a few entries in the array. 
- When changing the model to use full-float outputs, the model will output a more consistent array of predictions, but they seem to be independent of the input (ie. the output is always similar even though the input has changed to be another category). The output array of the C++ application also doesn't reflect the output array of the model when it is loaded by Python.  
- When loading the hello world model into the application using the same setup function, the model seems to produce a reasonable amount of error.

njeffrie, 2021-08-04:
Can you run the converted TFLite model using the [TFLite python interpreter](https://www.tensorflow.org/api_docs/python/tf/lite/Interpreter) and verify outputs?

duncmac16, 2021-08-04:
Well the input values are from a constant array generated by printing the values from a numpy array to a cpp file. The input values are also preprocessed in the training script. I also ran the TFLite python interpreter on all of the testing and training data and got an accuracy of about 94%. 

Another thing I considered while troubleshooting is the ordering of the numpy array, so I tried transposing the input, but it gives a very similar output

advaitjain, 2021-08-19:
Similar to https://github.com/tensorflow/tflite-micro/issues/438#issuecomment-902158126, our recommendation would be to try to debug on x86 first.

I'm going to close the current issue but please create a new one and tag the ESP32 maintainer (@vikramdattu) if the problem is isolated to ESP32.

closed 2021-08-19 by advaitjain