tensorflow/tflite-micro #341: No rule to make target...

straightupdeliciousfunk, 2021-07-30:
### System information

- Host OS Platform and Distribution: Ubuntu 20.04.2 LTS
- TensorFlow installed from (source or binary): source
- Tensorflow version (commit SHA if source): [e87305](https://github.com/tensorflow/tflite-micro/commit/e87305ee53c124188d0390b1ef8ec0555760d4d6)
- Target platform (e.g. Arm Mbed OS, Arduino Nano 33 etc.): Apollo3 (Sparkfun_Edge)

### Describe the problem
Running
`$ make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge hello_world`
or 
`make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge micro_speech_bin` from [Google codelabs](https://codelabs.developers.google.com/codelabs/sparkfun-tensorflow/#3)

results in the error:
`make: *** No rule to make target 'hello_world'.  Stop.`

No matter what I try, there doesn't seem to be a target to build any of these examples in the makefile. Per issue [#295](https://github.com/tensorflow/tflite-micro/issues/295) I've tried the [suggestion](https://github.com/tensorflow/tflite-micro/issues/295#issuecomment-882930759) from the comments, and the first two (clean & download) both work great, but again... there's no target for hello_world, micro_speech, etc. 

Help?

njeffrie, 2021-08-04:
Odd, I am able to run those commands in my TFLM repo with no issues.

Can you try re-cloning the TFLM repo `git clone https://github.com/tensorflow/tflite-micro.git` and running the same build in the newly cloned repo?

advaitjain, 2021-08-19:
We are in the process of moving hardware specific code into stand-alone repositories (https://github.com/tensorflow/tflite-micro/issues/407).

Please try the sparkfun edge repository at https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples

The following command will build all the examples:
```
make examples
```

closed 2021-08-19 by advaitjain

arun63pajith, 2023-12-02:
can anyone help me with this error

BlackPearl:~/tflite-micro$ make -f tensorflow/lite/micro/tools/make/Makefile generate_projects

tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/kissfft already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.

make: *** No rule to make target 'generate_projects'.  Stop

rascani, 2023-12-04:
The `generate_projects` target no longer exists.

c-z-c-2, 2024-01-25:
If you are following the [tensorflow document "Understand the C++ library"(https://www.tensorflow.org/lite/microcontrollers/library) the examples are located in tensorflow/lite/micro/examples, you may find [the readme in this page](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/examples/hello_world) useful.