tensorflow/tflite-micro #346: How do we include a tflite micro in a empty project?

stanleycelestin1, 2021-08-02:
I've worked through the examples successfully and wanted to work on my own computer vision projects using tf lite micro. My current solution is to use the person detection example and hack it out for my situation. 

Is there a standard way to get the required third party libraries like flatbuffers, ruy, and gemmlowp and a simplified tensorflow folder?

njeffrie, 2021-08-05:
have you tried project generation? For example `make -f tensorflow/lite/micro/tools/make/Makefile generate_hello_world_make_project`

That should generate a project in `tensorflow/lite/micro/tools/make/gen/<target>/prj/`.

njeffrie, 2021-08-05:
edit: I suggest using `tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py`

petewarden, 2021-08-23:
Closing now, unless there are any updates?

closed 2021-08-23 by petewarden