tensorflow/tflite-micro #347: fatal error: 'am_bsp.h' file not found

amitkac, 2021-08-02:
While compiling the hello_world code for development machine, I keep getting the following error.

`tensorflow/lite//micro/sparkfun_edge/system_setup.cc:22:10: fatal error: 'am_bsp.h' file not found`

Wondering if it is just me or is it a known bug.

resorcap, 2021-08-03:
I meet this problem on my mac too.  Centos is OK.

bychia, 2021-08-03:
same here.
macos 11.4.
GNU Make 4.3
Built for x86_64-apple-darwin20.1.0

njeffrie, 2021-08-03:
Can you try:
```
make -f tensorflow/lite/micro/tools/make/Makefile clean_downloads
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=sparkfun_edge third_party_downloads
```

See: https://github.com/tensorflow/tflite-micro/issues/295

amitkac, 2021-08-04:
Thanks for the reply @njeffrie .  Now, I am getting this error after running the second command,

`tensorflow/lite//micro/kernels/xtensa/softmax_int8_int16.cc:285:20: error: redefinition of 'TfLiteRegistration tflite::Register_SOFTMAX_INT8_INT16()`

bychia, 2021-08-04:
Same mentioned issue (Register_SOFTMAX_INT8_INT16()) as @amitkac 
btw, the exact command works on my Linux + make 4.2.1.
`make -f tensorflow/lite/micro/tools/make/Makefile test_hello_world_test`

vikramdattu, 2021-08-05:
The issue is because wrong files getting added to sources.

For me this change fixes the issue: https://github.com/tensorflow/tflite-micro/pull/363/commits/4ca3c0b4d6ba40f75585ef44b1231a1c61c0f367

amitkac, 2021-08-05:
Thank you for the suggestion @vikramdattu ,it worked. So this is what I did,

- Cloned fresh `tf-lite` repo as there have been some updates lately
- Update the `Makefile `as suggested by @vikramdattu 
- run commands as suggested by @njeffrie 
- Finally `make` the `test_hello_world_test` and `hello_world`, and check the output. It worked :)

njeffrie, 2021-08-05:
I've landed https://github.com/tensorflow/tflite-micro/pull/367 in order to fix this issue. Tip of tree should work now.

closed 2021-08-05 by njeffrie

amitkac, 2021-08-05:
Just checked it now, It works fine with the new fix to `Make` :) Thank you @njeffrie.