tensorflow/tflite-micro #350: tflite micro without any of the other stuff?

stanleycelestin1, 2021-08-03:
Is it possible to get the tensorflow folder without any of the board specific things, examples, etc to end up with just the simple tensorflow lite micro library?

njeffrie, 2021-08-03:
I suggest trying Make project generation:
```
make -f tensorflow/lite/micro/tools/make/Makefile generate_hello_world_make_project
```

This will generate a standalone Makefile project under `tensorflow/lite/micro/tools/make/gen/<host system>/prj/hello_world/make`

You can then modify / replace the hello_world example with your own code, or delete it entirely if you just wish to copy core TFLM files into your own project.

Alternatively, you can build libmicrolite.a to link into your project by running:
```
make -f tensorflow/lite/micro/tools/make/Makefile microlite
```
This will generate libmicrolite.a under `tensorflow/lite/micro/tools/make/gen/<host system>/lib/libtensorflow-microlite.a`

njeffrie, 2021-08-03:
One more alternative would be to use `tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py` directly.

advaitjain, 2021-08-03:
Please see https://github.com/tensorflow/tflite-micro/issues/332#issuecomment-888546457 for more details on the Python project generation script which is the recommended way going forward.

closed 2021-08-03 by advaitjain