tensorflow/tflite-micro #356: Make some of the private members of MicroInterpreter protected so derived classes can access them

andrewstanfordjason, 2021-08-05:
I would like some of the members of the MicroInterpreter class to be protected rather than private as I would like to write a derived class that has access to the allocations with graph_. In the past we(Xmos) had a derived class of the MicroInterpreter that implemented timing hooks (function calls that profiled the execution time of each operator) by patching the registrations, maybe there is a better way of doing this? Thanks

njeffrie, 2021-08-05:
Have you tried supplying a MicroProfiler when you create your interpreter? By default, if a profiler is supplied, it will be updated with profiling information tracking time / cycles elapsed to invoke each operator. See `tensorflow/lite/micro/benchmarks/person_detection_benchmark.cc` for an example using the MicroProfiler class. MicroProfiler methods are intended to be overridden to implement platform-specific timing code.

If that is not sufficient, I encourage you to look at the implementations in micro_graph.cc, since that's where I've personally inserted timing hooks most frequently. I suspect you're maintaining a fork of TFLM, so forking micro_graph.cc could be easier than the interpreter itself.

petewarden, 2021-08-23:
Closing, unless there are any updates @andrewstanfordjason ?

closed 2021-08-23 by petewarden

andrewstanfordjason, 2021-08-28:
No, the correct way to do it was to use the MicroProfiler. Thanks