tensorflow/tflite-micro #381: mbed_makefile.inc missing

swapnilsayansaha, 2021-08-08:
I am trying to compile programs for Mbed supported boards. Sample command:

```
make -f tensorflow/lite/micro/tools/make/Makefile TARGET=mbed hello_world
```
I get the following error:
```
tensorflow/lite/micro/tools/make/Makefile:564: tensorflow/lite/micro/tools/make/targets/mbed_makefile.inc: No such file or directory
make: *** No rule to make target 'tensorflow/lite/micro/tools/make/targets/mbed_makefile.inc'.  Stop.
```

It seems like the file in question is missing in tensorflow/lite/micro/tools/make/targets/ directory.

petewarden, 2021-08-23:
Closing, unless there are any updates.

closed 2021-08-23 by petewarden

swapnilsayansaha, 2021-08-23:
Used TF2.5.0's TFLiteMicro, which has the mbed make file and associated files.