tensorflow/tflite-micro #382: AllOpsResolver error error: expected class-name before '{' token while creating latest build for esp32.

AdityaJain1030, 2021-08-09:
I just built the latest ESP32 build with `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp generate_hello_world_esp_project`, and when trying to compile my project, the compiler throws and error saying ```/tensorflow/lite/micro/kernels/all_ops_resolver.h:22:54: error: expected class-name before '{' token```. My guess was that the bug had to do with not defining the `tOpCount` template, and the compiler couldn't detect the class name because of it, but when I changed 
```cpp
class AllOpsResolver : public MicroMutableOpResolver {
 public:
  AllOpsResolver();

 private:
  TF_LITE_REMOVE_VIRTUAL_DELETE
};```
to 
```cpp
class AllOpsResolver : public MicroMutableOpResolver<128> { <-- added tOpCount 
 public:
  AllOpsResolver();

 private:
  TF_LITE_REMOVE_VIRTUAL_DELETE
};```
it now throws the new error 
```
C:\Users\adity\.platformio\lib\EloquentTinyML\src/tensorflow/lite/micro/kernels/all_ops_resolver.h:22:53: error: expected '{' before '<' token
C:\Users\adity\.platformio\lib\EloquentTinyML\src/tensorflow/lite/micro/kernels/all_ops_resolver.h:22:53: error: expected unqualified-id before '<' token
C:\Users\adity\.platformio\lib\EloquentTinyML\src\tensorflow\lite\micro\kernels\all_ops_resolver.cpp:23:32: error: invalid use of incomplete type 'class tflite::ops::micro::AllOpsResolver'
```
Has anyone else encountered this error, and if so, know how to fix it?

rockyrhodes, 2021-08-11:
The current code in the main branch contains the "128" template parameter as you show here.  Maybe your branch is not up to date?

AdityaJain1030, 2021-08-12:
Ill try recloning and rebuilding soon

advaitjain, 2021-08-19:
I'm going to close this issue. Please open a new one and tag @vikramdattu (ESP32 maintainer) if you continue to see issues.

closed 2021-08-19 by advaitjain