tensorflow/tflite-micro #395: person_detection_test is not testing no_person_image and fails when doing so

mocleiri, 2021-08-12:
I created a micropython implementation of the person_detection example.  

I picked up the person_detection_model, the example image with person data and without person data from tflite-micro/tensorflow/lite/micro/tools/make/downloads/person_model_int8.

I converted the _person_image_data.cc, no_person_image_data.cc_ and _person_detect_model_data.cc_ back from byte arrays into binary files.

I made a python implementation like _person_deletection_test.cc_ where it runs inference on the person image and then on the no person image.

In micropython it works properly to classify the **person** but fails to classify the **no_person** image data.

```
~/git/tensorflow-micropython-examples/examples/person_detection$ /mnt/c/Users/Mike/Downloads/micropython ./person_detection.py
interpreter_make_new: model size = 300568, tensor area = 139264
Classify No Person Image
setup 9612 bytes on the inputTensor.
**'not a person' = -25, 'person' = 25**
Classify Person Image
setup 9612 bytes on the inputTensor.
'not a person' = -113, 'person' = 113
```

What is supposed to happen is that for no_person_image_data the **not a person** category is supposed to have a value larger that the **person** category but in this case the size of the number is smaller but the relationship doesn't invert as expected.

I found out how to run the _person_detection_test.cc_ and observed a totally different score for the no person image:

```
~/git/tensorflow-micropython-examples/tensorflow$ make -f tensorflow/lite/micro/tools/make/Makefile test_person_detection_test_int8
find: ‘../google/’: No such file or directory
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/person_model_int8 already exists, skipping the download.
tensorflow/lite/micro/tools/make/gen/linux_x86_64_default/bin/person_detection_test_int8 '~~~ALL TESTS PASSED~~~' linux
Testing TestInvoke
person data. person score: 113, no person score: -113

**no person data. person score: -72, no person score: 72**

Ran successfully

1/1 tests passed
```

Then I read the code and for the no person image its not actually using the image but instead doing a memset 0 which I think is zeroing all the bytes on the input tensor.

After changing the C code to run inference on the no_person_image_data array I get a failure and the same results as I saw in micropython:

```
~/git/tensorflow-micropython-examples/tensorflow$ make -f tensorflow/lite/micro/tools/make/Makefile test_person_detection_test_int8
find: ‘../google/’: No such file or directory
tensorflow/lite/micro/tools/make/downloads/flatbuffers already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/pigweed already exists, skipping the download.
tensorflow/lite/micro/tools/make/downloads/person_model_int8 already exists, skipping the download.
tensorflow/lite/micro/tools/make/gen/linux_x86_64_default/bin/person_detection_test_int8 '~~~ALL TESTS PASSED~~~' linux
Testing TestInvoke
person data.  person score: 113, no person score: -113

**no person data.  person score: 25, no person score: -25**

no_person_score > person_score failed at tensorflow/lite/micro/examples/person_detection/person_detection_test.cc:131
Ran successfully

0/1 tests passed
~~~SOME TESTS FAILED~~~

make: *** [tensorflow/lite/micro/examples/person_detection/Makefile.inc:53: test_person_detection_test_int8] Error 1
```
The code change I made to cause the test to fail is located here:
https://github.com/mocleiri/tflite-micro/commit/47f7be62961afd43df931001c722a704a8c1f264

Images are viewable here: [person_detection_images.zip](https://github.com/tensorflow/tflite-micro/files/6972564/person_detection_images.zip)

Download, unzip and run show-images.py and it will display both images.

mocleiri, 2021-08-15:
[image]

These are the example images.
| Image | Person Score | No Person Score | Notes|
| --- | ---| --- | --- |
| Person | 113 | -113 | As expected because person score > no person score |
| No Person | 25 | -25 | Unexpected because this is known not to be a person but person score > no person score. |
| memset(0) | -72 | 72 |  Expected since all zero's are not a person.  Score also matches expectations where no person score > person score. |

I was able to reproduce the memset(0) scores by setting the input image to zero's:

under line 38 of [person_detection.py](https://github.com/mocleiri/tensorflow-micropython-examples/blob/29f3375f41b36018a8767faa7e637b0ac5180135/examples/person_detection/person_detection.py#L38) replace the file loading and instead zero the bytes:
```python
for i in range (0, len(test_image)):
    test_image[i] = 0
```

mocleiri, 2021-08-19:
I think the main thing this issue is suggesting is that if you don't use the no_person_image_data then why is it being generated by the tooling. 

But probably its just that a different picture should be selected and scaled down to replace the existing no_person_image data.  

The new selection should work right with the model and classify with the no person category with a higher value then the person category.

advaitjain, 2021-08-20:
Thanks for debugging this issue.

For completeness, we do have an internal issue to fix this, but it is not currently being actively worked on:
https://github.com/tensorflow/tflite-micro/blob/a9f2e03b943990bc958ce3b92ca76ffc27fcf6d6/tensorflow/lite/micro/examples/person_detection/person_detection_test.cc#L104-L105

closed 2021-09-08 by njeffrie