tensorflow/tflite-micro #396: Quantize does not support uInt8 inputs

AdityaJain1030, 2021-08-12:
I have a quantized version of mobilenet, quantized with the following code
```py
def representative_dataset():
  filename_dataset = tf.data.Dataset.list_files("/content/imagenet-sample-images/*.JPEG")
  for i in range(100):
    image = next(iter(filename_dataset))
    image = tf.io.read_file(image)
    image = tf.io.decode_jpeg(image, channels=3)
    image = tf.image.resize(image, [320, 240])
    image = tf.cast(image / 255., tf.float32)
    image = tf.expand_dims(image, 0)
    yield [image]

converter = tf.lite.TFLiteConverter.from_keras_model(m)

converter.optimizations = [tf.lite.Optimize.DEFAULT]
converter.representative_dataset = representative_dataset
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8]
converter.inference_input_type = tf.uint8  # or tf.uint8
converter.inference_output_type = tf.uint8  # or tf.uint8
tflite_model = converter.convert()

with open('model.tflite', 'wb') as f:
  f.write(tflite_model)
```
When I convert 
```py
converter.inference_input_type = tf.uint8  # or tf.uint8
converter.inference_output_type = tf.uint8  # or tf.uint8
``` 
to 
```py
converter.inference_input_type = tf.int8  # or tf.uint8
converter.inference_output_type = tf.int8  # or tf.uint8
```
My code works fine with the tflite micro library. However, for some reason, converting the input to int8 also hurts the model accuracy a bit, whereas uint8 doesn't seem to do so. But by using uint8 input tensors, I run into this issue with TFMicro
```
lib\tfmicro\tensorflow\lite\micro\kernels\quantize.cc:69 input->type == kTfLiteFloat32 || input->type == kTfLiteInt16 || input->type == kTfLiteInt8 was not true.
Node QUANTIZE (number 0f) failed to prepare with status 1
```
Is there any way to get around this issue, or do I have to stick with the int8s? Also if I do have to use int8s only, is there a way to get the same accuracy I've been getting with uint8s?

AdityaJain1030, 2021-08-12:
Just saw https://github.com/tensorflow/tflite-micro/issues/280, but that still leaves me curious as to why the tf.uint8 model still works with a higher accuracy in regular tflite.

closed 2021-08-12 by AdityaJain1030

advaitjain, 2021-08-12:
Please feel free to email the tflite mailing list for questions about quantization.

https://github.com/tensorflow/tflite-micro#getting-help