tensorflow/tflite-micro #400: Reshape ops is raised an error "num_input_elements != num_output_elements" (50 != 1)

colynhn, 2021-08-13:
When I replace my own model to the hello world model，the reshape ops is raised an error "num_input_elements != num_output_elements" (50 != 1) when input is a 2D tensor and output is a 1D tensor.

`reshape input: int32[1,50]`
`reshape output: int32[1]`
`reshape shape value = -1`

 But when I trace to the code: `tensorflow/lite/micro/kernels/reshape.cc: 56 int value = output_shape->data[i];` the value is always 1

So how can i fix this? thanks.
[image]
[image]

resorcap, 2021-08-17:
I met this problem too.

petewarden, 2021-08-23:
Can you share a complete script you're using to create the model?

petewarden, 2021-10-04:
Closing for now since there's been no update in a while. Please do reopen if you have more information available.

closed 2021-10-04 by petewarden