tensorflow/tflite-micro #407: Move hardware / platform support to repositories that are external to tflite-micro

advaitjain, 2021-08-13:
As an extension of https://github.com/tensorflow/tensorflow/issues/47413, we would like to move hardware platform support and platform-specific customization of the examples out of the tflite-micro repository and into platform specific repositories.

The goal with such a transition is:
 * Give more control over the TFLM integration to the maintainers of the specific platform, including set up of the IDE / build system etc.
 * Allow platform maintainers to decide at what cadence they would like to pull in updates from TFLM, make releases ...
 * Allow platform maintainers and easier way to customize and document the TFLM examples for their specific target.
 * Reduce the amount of code in the TFLM repository that can not be tested or maintained by the TFLM maintainers.
 * Simplify the examples in the TFLM repository (by not attempting to support various hardware variations, peripherals ...)

We are going to accelerate this transition by starting to delete the existing platform support because this is a common pain point and the existing way of supporting various hardware platforms is error-prone and not scalable. Some related discussions on this topic:
 * https://github.com/tensorflow/tflite-micro/issues/332#issuecomment-888436216
 * https://github.com/tensorflow/tflite-micro/issues/294#issuecomment-883922085
 * https://github.com/tensorflow/tflite-micro/issues/264#issuecomment-877307077
 * https://github.com/tensorflow/tflite-micro/issues/246#issuecomment-875958553
 * https://github.com/tensorflow/tflite-micro/issues/350
 * https://github.com/tensorflow/tflite-micro/issues/170#issuecomment-860888987

advaitjain, 2021-08-13:
Here is a (partial) list of existing integrations that will be affected by this change along with my guess of some of the relevant stakeholders:

 * apollo3evb, sparkfun_edge: @oclyke
 * arc: @gerbauz
 * arduino: migrated to https://github.com/tensorflow/tflite-micro-arduino-examples
 * ceva: @yair-ehrenwald
 * ecm3531: \<unknown\>
 * esp: @vikramdattu
 * himax: @bigcat-himax
 * mbed: @freddan80 
 * openmvcam: @kwagyeman
 * riscv32_mcu: @petewarden 
 * spresense: @takayoshi-k
 * stm32f4hal: @alxhoff
 * xcore: @andrewxcav

advaitjain, 2021-08-14:
Updated recommendation for how to integrate TFLM with a specific platform is at https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/docs/new_platform_support.md

vikramdattu, 2021-08-16:
@advaitjain the new platform specific project need not be part of TensorFlow repo?

yair-ehrenwald, 2021-08-16:
@advaitjain What exactly does this refer to, just the platform specific code under the examples directory or also optimized kernels etc?

advaitjain, 2021-08-16:
> @advaitjain the new platform specific project need not be part of TensorFlow repo?

Yes. We would like to have the existing and new platform specific be external to the TFLM repository.

advaitjain, 2021-08-16:
> @advaitjain What exactly does this refer to, just the platform specific code under the examples directory or also optimized kernels etc?

This only refers to platform specific code under the examples as well as project generation support from within the Makefiles.

The TFLM repository will continue to be a central location for optimized kernels. In fact, with the new GitHub Actions based CI, we are now set up to have the optimized kernel implementations with proprietary toolchains also be tested with the TFLM CI -- happy to talk more about that if it is of interest to you.

embeddetech, 2021-08-18:
Hi @advaitjain,
Will I still be able to do a pull request at some point for my Visual Studio compatibility changes? I am definitely interested in creating a Virtuoso TFLM repository branch, however to be able to continue getting latest for periodic synchronization with the main TFLM repo it would sure be nice to be in sync with my changes (may not even be practically possible to do a repo unless we're in sync first?)

Thanks!

TI-AviBerko, 2021-08-19:
@advaitjain 

1.	You’re advising setting up a github sync action to sync tflite with the vendor’s example repo, but that doesn’t include the third party code that is only downloaded when using the “create tree” script – is this correct?
2.	Concerning CI, I’m wondering how long/expensive the Arduino/edge CI’s are? Setting up our toolchain can take a bit

advaitjain, 2021-08-19:
Good questions.

> 1. You’re advising setting up a github sync action to sync tflite with the vendor’s example repo, but that doesn’t include the third party code that is only downloaded when using the “create tree” script – is this correct?

Exactly how you want to handle the third_party_dependencies is your choice. The create_tflm_tree.py script does put the third_party dependencies for TFLM into a `third_party` directory. For libraries like flatbuffers, ruy and gemmlowp, I would guess that most integrations would want to use exactly what TFLM has.

For libraries that may be more broadly used (e.g. CMSIS and kissfft), the hardware maintainers will want to make a choice on a case-by-case basis.

As an example, for the Arduino we are using all the third_party code as-is:
https://github.com/tensorflow/tflite-micro-arduino-examples/tree/main/src/third_party

FOr the sparkfun_edge, [additional CMSIS sources](https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples/tree/main/third_party/cmsis/CMSIS/DSP/Source) are needed specifically for the board bringup and so we maintain those separately but still copy over all the TFLM 3rd party dependencies:
https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples/blob/0cc8217949d3069d32187c2f2bb62aaedd73c669/scripts/sync_from_tflite_micro.sh#L55

> 2. Concerning CI, I’m wondering how long/expensive the Arduino/edge CI’s are? Setting up our toolchain can take a bit

Again, the choice is yours though here are some of the options.

You can see the duration of the CI at these links:
 * https://github.com/tensorflow/tflite-micro-arduino-examples/actions/workflows/ci.yml
 * https://github.com/advaitjain/tflite-micro-sparkfun-edge-examples/actions/workflows/ci.yml

If the toolchain setup is time consuming enough that you'd rather not repeat it for each CI run, GitHub has good support to do the installation once on a docker container.

Here is one example where TFLM uses a docker container as part of CI:
https://github.com/tensorflow/tflite-micro/blob/c62b3bd94a4ee8534e8b43eb2757766eca5f081c/.github/workflows/ci.yml#L72-L78

gerbauz, 2021-09-08:
Hi @advaitjain, I don't quite understand - what will happen to "microlite" target, when tflm is generated as a static library. Is this case is somehow supported by new build system, or it will continue to be supported by old make build system?

advaitjain, 2021-09-09:
The TFLM Makefiles will continue to be used to build the `microlite` target. And for any target with optimized kernel implementations (such as arc), the TFLM repo will continue to be a central location for the optimizations.

In fact, as much as possible we would be open to hosting CI for these optimized kernel implementations as part of the TFLM GitHub CI.

What we do want is
 * move all the target specific code in the examples directory (such as the arc and arc_emdsp folders in micro_speech, hello_world etc) to a separate standalone examples repository (say https://github.com/foss-for-synopsys-dwc-arc-processors/tflite_micro_arc_examples)
 *  remove the support for Makefile based project generation (i.e. no longer support targets such as `generate_person_detection_test_int8_make_project` since once it is set up, the arc-specific examples directory should have the code ready for use.

Please feel free to create a new github issue that is specific to ARC and we would be happy to work with you to figure out how to make this work.

Lachheeldedag, 2021-10-01:
Hi I'm trying to implement the TFLM library on a cortex m-33 processor. I created the tflm_tree and the static library. When I try to compile using GNU-ARM Cross c++ compiler I get many error messages: template with C linkage; template specialization with C linkage and conflicting declaration of C function. Is this a known problem? I suspect it has something to do with the flatbuffer files, as many error messages seem to come from that direction.

advaitjain, 2021-11-01:
> Hi @advaitjain,
> 
> I'm now working on ARC target to meet all requirements. My question is: Can we keep this directory [`main`/tensorflow/lite/micro/tools/make/targets/arc](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/tools/make/targets/arc?rgh-link-date=2021-10-18T09%3A14%3A35Z) with make files inside? They are actually required to build static lite micro library.
> 

Responded to this at https://github.com/tensorflow/tflite-micro/issues/637#issuecomment-956572075

> Also, what's the fate of the kernel tests? As far as I remember, they also used to be generated using legacy project generation.

The kernel tests will not be part of the project generation.

Instead, we are going to trend towards having accuracy test-data suites which will have kernel parameters, golden inputs and outputs that can be integrated with external build systems / testing harnesses etc.

We do not have a specific timeline for this accuracy test-data suite, and the current expectation is that the kernel tests will be run from the TFLM repository.

vikramdattu, 2021-11-08:
Dear all,

Please find esp-examples repo here: https://github.com/espressif/tflite-micro-esp-examples

challiwill, 2022-05-25:
Some of the docs still need to be updated to reflect changes around this, such as https://www.tensorflow.org/lite/microcontrollers/library?authuser=2#generate_projects_for_other_platforms still references the `generate_projects` make command, which was removed in https://github.com/tensorflow/tflite-micro/commit/2620f729b7e37a3383cb80af99cc23bbafc5a7ec

kwagyeman, 2023-06-30:
This still relevant?

advaitjain, 2023-07-12:
There are likely loose ends around documentation, but the code changes are complete.

closed 2023-07-12 by advaitjain