tensorflow/tflite-micro #409: Cortex-M Corstone 300 (FVP) build is failing

advaitjain, 2021-08-13:
https://github.com/tensorflow/tflite-micro/pull/337 broke the FVP build with the following error message:

```
tensorflow/lite/micro/tools/make/downloads/person_model_int8/person_detect_model_data_vela.cc:2:10: fatal error: tensorflow/lite/micro/examples/person_detection/person_detect_model_data.h: No such file or directory
    2 | #include "tensorflow/lite/micro/examples/person_detection/person_detect_model_data.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [tensorflow/lite/micro/tools/make/Makefile:599: tensorflow/lite/micro/tools/make/gen/cortex_m_corstone_300_cortex-m55_default/obj/core/tensorflow/lite/micro/tools/make/downloads/person_model_int8/person_detect_model_data_vela.o] Error 1
make: *** Waiting for unfinished jobs....
tensorflow/lite/micro/examples/network_tester/network_tester_test.cc:27:10: fatal error: tensorflow/lite/micro/examples/person_detection/person_detect_model_data.h: No such file or directory
   27 | #include "tensorflow/lite/micro/examples/person_detection/person_detect_model_data.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

advaitjain, 2021-08-13:
@mansnils, would you be willing to make a fix for this.

We are now using the tflite files as the source of truth and the file is in the repo at https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/micro/models

What will need to be done (I think) is to have the vela conversion put the converted .cc and header in this new location:
```
tensorflow/lite/micro/tools/make/gen/cortex_m_corstone_300_cortex-m55_default/genfiles/tensorflow/lite/micro/models
```

mansnils, 2021-08-16:
It seems the problem is caused by this file is now removed:
./tensorflow/lite/micro/examples/person_detection/person_detect_model_data.h

But the same file is available here:
./tensorflow/lite/micro/tools/make/downloads/person_model_int8/person_detect_model_data.h

So I will just update the path in the network example.

closed 2021-08-16 by advaitjain