tensorflow/tflite-micro #423: Person_detection_int8 example with Mbed is broken

mansnils, 2021-08-17:
How to reproduce:
make -f tensorflow/lite/micro/tools/make/Makefile OPTIMIZED_KERNEL_DIR=cmsis_nn generate_person_detection_int8_mbed_project
cd tensorflow/lite/micro/tools/make/gen/linux_x86_64_default/prj/person_detection_int8/mbed/
mbed new .
mbed compile -m DISCO_F746NG -t GCC_ARM

Compile [ 58.4%]: main_functions.cc
[Fatal Error] main_functions.cc@24,67: tensorflow/lite/micro/models/person_detect_model_data.h: No such file or directory
[ERROR] ./tensorflow/lite/micro/examples/person_detection/main_functions.cc:24:67: fatal error: tensorflow/lite/micro/models/person_detect_model_data.h: No such file or directory
 #include "tensorflow/lite/micro/models/person_detect_model_data.h"

Problem is that person_detect_model_data.h is now expected to be found in tensorflow/lite/micro/models/.
This file is now generated for Bazel builds but not for this Mbed build.

petewarden, 2021-08-23:
This looks closely related to https://github.com/tensorflow/tflite-micro/issues/458.

njeffrie, 2021-09-09:
Try to use create_tflm_tree.py to generate the project rather than makefile based project generation.
```
python3 tensorflow/lite/micro/tools/project_generation/create_tflm_tree.py  --makefile_options OPTIMIZED_KERNEL_DIR=cmsis_nn --examples person_detection <output_dir> 
cd <output_dir>
mbed new .
mbed compile -m DISCO_F746NG -t GCC_ARM -D CMSIS_NN

```
Note: I had to:
```
cd mbed-os && mbed update master
```

closed 2021-09-09 by njeffrie